arkypita/LaserGRBL #1: Adding preview images

arkypita, 2016-12-04:
[image]
[image]

closed 2016-12-04 by arkypita

arkypita, 2016-12-20:
[image]

arkypita, 2016-12-20:
[image]

arkypita, 2016-12-20:
[image]

arkypita, 2016-12-22:
[image]

Rost3, 2016-12-25:
i have a Eleks Laser 5.5W It burns all the coals at grayscale, too powerful. can you add a power limits. min power and max power to grayscale.

arkypita, 2016-12-25:
Yes, it's a good idea! Since the software is in early development, the program is very wired on my hardware configuration, with limited configuration options. One of my next goals is to add a configuration page that allows you to set some parameters.

However your request is very sensible, so I will provide to implement it as soon as possible in Raster Import Form.

Greetings

arkypita, 2016-12-25:
@Rost3 Check version 1.0.2 on https://github.com/arkypita/LaserGRBL/releases/tag/v1.0.2
You can find an option to scale laser power

Rost3, 2016-12-25:
is good!!! some program have curves for scale laser power. but is for future.
[image]
[image]
I used to calibrate power the gray gradient

Rost3, 2016-12-25:
at the beginning  of a line is switched on the laser, burns through the point and then moves. it might be worth to make so that the laser worked only when moving?

arkypita, 2016-12-26:
@Rost3 Thank for your feedback!
I suggest you to open new issues instead of posting at the bottom of this topic. To do that, simply click the green button "new issue" on top right of the page https://github.com/arkypita/LaserGRBL/issues
Use one issue for any different issue, this help me to keep track of opened request.

> is good!!! some program have curves for scale laser power. but is for future.
It's a nice suggestion, could be a future goal. (not to difficult to implement)

> at the beginning of a line is switched on the laser, burns through the point and then moves. it might be worth to make so that the laser worked only when moving?

This is because at the beginning of the movement laser is firm, then rapidly accelerate to travel speed.
During the initial acceleration speed goes from zero to travel speed in milliseconds, but defined power still remain the same so at the beginning of the acceleration, when speed is low, laser burn more.
Since acceleration are managed by onboard software (grbl on arduino) the best approach is to scale realtime laser power according to realtime speed.

The last version (1.1) https://github.com/gnea/grbl/wiki of grbl for arduino can do that: 
> Dynamic Laser Power Scaling with Speed : If your machine has low accelerations, this option will automagically scale the laser power based on how fast Grbl is traveling, so you won't have burnt corners when your CNC has to make a turn! Enabled by the M4 spindle CCW command when laser mode is enabled.

If you decide to upgrade your arduino firmware to 1.1 please be sure to backup your settings ($$) because the update will reset arduino memory. Also keep a copy of config.h to restore the same macro options.

Rost3, 2016-12-26:
i use grbl 1.1. thanks for the explanation of the grbl settings. Your program best I've seen.

arkypita, 2017-02-27:
[image]
[image]

arkypita, 2017-02-27:
[image]

arkypita, 2017-03-04:
[image]