arkypita/LaserGRBL #2: New program's functions that may be useful

Rost3, 2016-12-26:
need to add "frame" around the pictures to accurately position the workpiece. laser not work, only go around the pictures. is my laser (not only my) has a "small power" button, do not burn, only shines.

Rost3, 2016-12-30:
4. creation G code from the vector dxf maybe from cdr.

arkypita, 2016-12-30:
Number 2 and 3 are easy to implement (diagonal is less quick to implement, but could be done too) but not really necessary. Number 1 is interesting and useful for anyone.

I will probabily implement them in the near future.

Number 4 is a professional feature that I don't need, and i will never need, so it's implementation is not a target for me. Development of LaserGRBL is driven by my needs, and claim to be simple and lightweight.
However is free and open software, so anyone with programming skill can fork and implement more function.

mayhem2408, 2016-12-30:
I run a CoreXY geometry rig and diagonal engraving produces the best results and allows the stepper motors to work in a similar way to a traditional XY Rig. Not a deal breaker, but diagonal work be great for some people.

mayhem2408, 2016-12-31:
I have forked the code and have implemented vertical engraving and working on 45 degree diagonal engraving.

mayhem2408, 2016-12-31:
My older rig has a single stepper on the X axis and 2 steppers for the Y axis. For some reason vertical scans seem a little smoother than horizontal. Never been able to figure out why, so always used vertical engraving. 
My newer rig uses CoreXY, so it does not have a stepper for X and a stepper for Y. 45 Degree engraving allows one stepper to be in hold position, while the other stepper is in motion. It produces great results with my python scripted code.

arkypita, 2017-01-01:
> I have forked the code and have implemented vertical engraving and working on 45 degree diagonal engraving.

Well done @mayhem2408!
I will insert your code into the main version, and i will release a new version.

arkypita, 2017-01-02:
@mayhem2408 You can now find H/V engraving in [lastest release] (https://github.com/arkypita/LaserGRBL/releases)

Diagonal is not implemented yet, but I have rewrited all code generation to be more virtual about X and Y, so i think make it working in diagonal way could be easy to do.

arkypita, 2017-01-06:
@mayhem2408 @Rost3 

Diagonal engraving now supported
https://github.com/arkypita/LaserGRBL/releases/tag/v1.3

arkypita, 2017-01-07:
Image vectorization now supported
https://github.com/arkypita/LaserGRBL/releases/tag/v2.0

mayhem2408, 2017-01-08:
@arkypita Diagonal is working great and vector is nice to. I'll post some photos of a test engrave very soon.

closed 2017-02-01 by arkypita