arkypita/LaserGRBL #3: laser GRBL config file

mkeyno, 2016-12-28:
sorry to open the ticket for my question , I was wondering if you have successfully implemented GRBL for actual  cnc laser machine, I really interested to know more about your experience

arkypita, 2016-12-29:
> sorry to open the ticket for my question

It's ok

>  I was wondering if you have successfully implemented GRBL for actual cnc laser machine, I really interested to know more about your experience

I'm not sure I understand the question.
However I have developed LaserGRBL for a personal use. After self-built a laser engraver, I tried to use different software without finding one that is suitable to my needs. Either they are too complex, or too simple.

And 'so the need arose and the pleasure of writing a simple, lightweight software, but it does everything I need. For now features include a very reliable GCode streamer on com port, and the loading a raster image and convert to GCode scan lines with PWM modulation.

I would soon add a conversion from raster to vector (image vectorialization) that allows to do the tracking edges.

My engraver is very homemade, made with a wooden frame and slides for drawers, bought for cheap at the hardware store. Now I'm currently updating it, with better mechanical created by the machine itself.

Unfortunately I do not have the opportunity to work with professional CNC machines, so I can not test my software environment different from that hobby (to which it is intended)

[image]
[image]
[image]

closed 2017-01-06 by arkypita

mkeyno, 2017-01-07:
thanks for reply , I mean if you've built  successfully  cnc laser base on GRBL firmware ,I really interested to know how to configure the lib and setup your hardware . if you know any links for that I really appreciated to share  it

arkypita, 2017-01-07:
Yes, and it is not particularly difficult. You need an Arduino one, a [arduino CNC shield](http://blog.protoneer.co.nz/arduino-cnc-shield/), and two/three driver chip for stepper motors.

Starting point could be: https://github.com/gnea/grbl/wiki/Connecting-Grbl

Google is your friend, and if you search www.google.it/search?q=grbl+cnc+arduino you can find a lot of DIY project with instructions.