arkypita/LaserGRBL #5: question

Rost3, 2017-01-03:
Can I run LaserGRBL on Linux? Raspberry Pi 3. my computer is too powerful (i7 ram 16GB) to use it in workshop. or better to buy a cheap old computer for it.

arkypita, 2017-01-03:
All the code of LaserGRBL is C# so I think it could be compiled for Linux in Mono, using MonoDevelop.
I have never done it before, and I have no linux machine at the moment, so I can't help you with this task. If you have skill you can tray yourself. In case, let me know if works

Jakeler, 2017-12-02:
It compiles fine on Arch Linux (xbuild) with latest mono 5.0.0. But when trying to run with `mono LaserGRBL.exe` i always get:
```
Unhandled Exception:
System.TypeInitializationException: The type initializer for 'Tools.TimingBase' threw an exception. ---> System.TypeInitializationException: The type initializer for 'Tools.HiResTimer' threw an exception. ---> System.EntryPointNotFoundException: QueryPerformanceFrequency
  at (wrapper managed-to-native) Tools.HiResTimer:QueryPerformanceFrequency (long&)
  at Tools.HiResTimer..cctor () [0x00025] in <4bb2d302ea56443d9efd6d6ed57ceec3>:0 
   --- End of inner exception stack trace ---
  at Tools.TimingBase..cctor () [0x00000] in <4bb2d302ea56443d9efd6d6ed57ceec3>:0 
   --- End of inner exception stack trace ---
  at LaserGRBL.Program.Main (System.String[] args) [0x00000] in <4bb2d302ea56443d9efd6d6ed57ceec3>:0 
[ERROR] FATAL UNHANDLED EXCEPTION: System.TypeInitializationException: The type initializer for 'Tools.TimingBase' threw an exception. ---> System.TypeInitializationException: The type initializer for 'Tools.HiResTimer' threw an exception. ---> System.EntryPointNotFoundException: QueryPerformanceFrequency
  at (wrapper managed-to-native) Tools.HiResTimer:QueryPerformanceFrequency (long&)
  at Tools.HiResTimer..cctor () [0x00025] in <4bb2d302ea56443d9efd6d6ed57ceec3>:0 
   --- End of inner exception stack trace ---
  at Tools.TimingBase..cctor () [0x00000] in <4bb2d302ea56443d9efd6d6ed57ceec3>:0 
   --- End of inner exception stack trace ---
  at LaserGRBL.Program.Main (System.String[] args) [0x00000] in <4bb2d302ea56443d9efd6d6ed57ceec3>:0
```
Can this be solved?

arkypita, 2017-12-02:
Hi Jackeler, I am happy that someone is trying to port LaserGRBL in linux.

TimingBase is a class for precise and hi-resolution timing. It is based on 
QueryPerformanceFrequency and QueryPerformanceCounter windows API and I suppose that those API are OS specific function and they are not available on mono. Fortunately it is not too difficult to replace them with something similar.

You can try to replace the whole HiResTimer class with this code:

```
	public static class HiResTimer
	{
		private const long NANO_IN_MILLI = 1000 * 1000;
		private static Stopwatch watch = new Stopwatch(); 

		static HiResTimer() //costruttore static, viene chiamato prima del primo utilizzo della classe
		{watch.Start();}

		public static long TotalMilliseconds
		{ get { return TotalNano / NANO_IN_MILLI; } }

		public static long TotalNano
		{
			get
			{
				return (long)((double)watch.ElapsedTicks / (double)Stopwatch.Frequency * 1000000000.0);
			}
		}

	}
```

Jakeler, 2017-12-02:
Thank you for the fast reply. That works!

Then i got this:
```Unhandled Exception:
System.DllNotFoundException: ntdll.dll
  at (wrapper managed-to-native) Tools.TaskScheduler:NtSetTimerResolution (uint,bool,uint&)
  at Tools.TaskScheduler.SetClockResolution (System.Int32 msec) [0x0000d] in <a1ec3de5fed24c81a1d181c135352908>:0 
  at LaserGRBL.Program.Main (System.String[] args) [0x00051] in <a1ec3de5fed24c81a1d181c135352908>:0 
[ERROR] FATAL UNHANDLED EXCEPTION: System.DllNotFoundException: ntdll.dll
  at (wrapper managed-to-native) Tools.TaskScheduler:NtSetTimerResolution (uint,bool,uint&)
  at Tools.TaskScheduler.SetClockResolution (System.Int32 msec) [0x0000d] in <a1ec3de5fed24c81a1d181c135352908>:0 
  at LaserGRBL.Program.Main (System.String[] args) [0x00051] in <a1ec3de5fed24c81a1d181c135352908>:0 
```
So i simply commented it out, like this:
```	
public class TaskScheduler
	{

		//[System.Runtime.InteropServices.DllImport("ntdll.dll", EntryPoint = "NtSetTimerResolution")]
		private static extern void NtSetTimerResolution(uint DesiredResolution, bool SetResolution, ref uint CurrentResolution);

		public static void SetClockResolution(int msec) //volendo potrebbe gestire anche 0.5 msec ma a noi basta 1-15 msec, quindi usiamo int
		{
			uint DesiredResolution = (uint)msec * 10000;
			bool SetResolution = true;
			uint CurrentResolution = 0;
			//NtSetTimerResolution(DesiredResolution, SetResolution, ref CurrentResolution);
		}

	}
```
Does this break something?

Anyway, i can start it now without direct error/crash on Linux! One optical issue is that the menu bar has no font, but function seems okay. I will now test if the connection to my machine works! :)

arkypita, 2017-12-02:
SetTimerResolution change the frequency of task- switching that in windows is about 66Hz (15 mSec) to a higher frequency that made LaserGRBL more responsive and fast in streaming g-code to arduino.

It's ok to comment it out

Jakeler, 2017-12-02:
Okay good, that makes sense.

Test results:

- Connection to GRBL 1.1 controller: works!
- Loading gcode file and sending to controller: works!
- "Type gcode here" field: fail/crash at any input:
```
System.NullReferenceException: Object reference not set to an instance of an object
  at LaserGRBL.UserControls.PlaceholderTextBox.waterMarkContainer_Paint (System.Object sender, System.Windows.Forms.PaintEventArgs e) [0x0000e] in <eb1043efb68f4c3593b70ed0451be259>:0 
  at System.Windows.Forms.Control.OnPaint (System.Windows.Forms.PaintEventArgs e) [0x00019] in <a98ec586cd3441aeb3d7919d9ad41ff2>:0 
  at System.Windows.Forms.Control.WmPaint (System.Windows.Forms.Message& m) [0x0007b] in <a98ec586cd3441aeb3d7919d9ad41ff2>:0 
  at System.Windows.Forms.Control.WndProc (System.Windows.Forms.Message& m) [0x001a4] in <a98ec586cd3441aeb3d7919d9ad41ff2>:0 
  at System.Windows.Forms.ScrollableControl.WndProc (System.Windows.Forms.Message& m) [0x00000] in <a98ec586cd3441aeb3d7919d9ad41ff2>:0 
  at System.Windows.Forms.Control+ControlWindowTarget.OnMessage (System.Windows.Forms.Message& m) [0x00000] in <a98ec586cd3441aeb3d7919d9ad41ff2>:0 
  at System.Windows.Forms.Control+ControlNativeWindow.WndProc (System.Windows.Forms.Message& m) [0x0000b] in <a98ec586cd3441aeb3d7919d9ad41ff2>:0 
  at System.Windows.Forms.NativeWindow.WndProc (System.IntPtr hWnd, System.Windows.Forms.Msg msg, System.IntPtr wParam, System.IntPtr lParam) [0x00085] in <a98ec586cd3441aeb3d7919d9ad41ff2>:0 
```
- Machine control (move): works!
- Reset+Lock Button: works!
- Create custom button and send the gcode: works!

arkypita, 2017-12-02:
Great job @jakeler

Comment DrawWaterMark in PlaceholderTextBox class.

protected override void OnPaint(PaintEventArgs e)
		{
			base.OnPaint(e);
			//Draw the watermark even in design time
			DrawWaterMark();
		}

Jakeler, 2017-12-03:
Had to comment it in `ThisHasFocus()` and `ThisTextChanged()` to avoid the error. It causes no problem in `OnPaint()` and `Initialize()`. I have the changes now in a fork: https://github.com/Jakeler/LaserGRBL/commit/1cea400d96600d561360c3c22a47d39224dd10e2

So i was able to successfully engrave something with LaserGRBL now. More test results:
- Settings, gbrl config, hotkeys: works
- Loading a raster image (png): Does nothing, no new window, no error...
- Speed, Power override popup: extremely small (unusable)
- UI Screenshot:
[image]
- Color schemes: background change works, but fonts stay grey, not readable in dark/hacker option

arkypita, 2017-12-03:
Hi @Jakeler 
I am very happy with your progress & results with mono. When I do my first experiments with an older version of LaserGRBL I could not go much further. My biggest problems were that the version of mono I was using was very old (3.2.8) and suffers several bugs that prevent winform program to function properly. I had done these tests on a raspberry, because I do not have a real linux PC available.

With Arch Linux (xbuild) and mono 5.0.0 i hope you can achieve better result then me. However i remember the biggest issue was on Raster Import form. Not because of my vectorizing/line2line code, but because the WinForm layout I am using.

I see some issue that should be the same you have now. I.e. the misalignment of the connect button, of the stop-continue button etc. The whole interface of LaserGRBL is designed using "TableLayoutPanel" that allow me to make it working very well on different resolution, resizing, different space used in different string translation etc.

Maybe some form should be re-designed to use a simpler & fixed layout schema to work in mono.

arkypita, 2017-12-03:
I should have an old laptop that I can sacrifice to a linux installation. I would be happy to help you with this job.

Jakeler, 2017-12-03:
According to to Mono docs, support for Windows Forms 2.0 is complete, but it seems not the same like on Windows. I also noticed that resizing the window smaller does not move the stop-continue buttons back, causes them to disappear at some point. Overall, most elements look not quite right. I would also prefer a more modern style, instead of this Windows 2000 theme on Linux.

I have not really any experience with C# and Windows GUI stuff, so i am not sure if i can do this or help you a lot. Now looking at the code i started to understand it a bit, but still... In addition some code is commented in Italian, which i don't understand. 
I can help you though with setting up Linux for development, testing and other related questions. When the software is working i could do proper packaging for Linux distributions, so that everone can install and update it easily.

You don't neccessarily need addidtional hardware, if you have a half decent PC you can run it just as virtual machine guest with VirtualBox. So you leave your Windows Host running and can quickly start another OS virtual. (I do this often the other way around, with Windows as guest on my Linux PC, especially for development very useful)

arkypita, 2017-12-04:
I am able to install linux both on physical or virtual machine, this is not a problem for me. The only problem is to find time :-)

In any case, I take advantage of your availability to obtain some information that can speed up this step.

What distro exactly you use? Could you provide me a link to an installable ISO?
Is the mono compiler or IDE already present in distro or you have to install later?

More: when I do my old try i remember that I cannot graphical edit my winform UI directly from monodevelop, because they only support editing GTK Form. Is the situation still the same? I mean, are you able to open a form, like mainform, and edit it graphically from monodevelop or not?

arkypita, 2017-12-04:
Update; i have installed Linux Mint with mono v5.4.16 and monodevelop 7.1. Both are the latest version.
I am stuck on a problem with resgen.exe as you can see in picture. Any hint?

[image]

arkypita, 2017-12-04:
Update: solved (switching to .net 4.5)

Jakeler, 2017-12-04:
Awesome to see that you are on it! I had not much time today to answer, anyway everything is solved now? Let me know how it goes and if you need help with something. ;)

arkypita, 2017-12-04:
You can find my result on https://github.com/arkypita/LaserGRBL under Linux branch
https://github.com/arkypita/LaserGRBL/tree/Linux

All the problem that I have found the last time I tried to do a linux version still exists.
Biggest issue is on tablelayout panel, that is not well supported. This broke all my layout.
Also some problem with image manipulation functions.

Now raster image form opens, but due to layout problem is unusable

arkypita, 2018-01-16:
I read this on internet about T2Laser under Linux.
T2Laser is a .NET executable like LaserGRBL is.

> T2Laser full working on linux.
> i  uploaded firmware and connect fine.
>
> steps.
> 
> uninstall wine and ppa
>
> sudo apt install ppa-purge
> sudo ppa-purge ppa:ubuntu-wine/ppa
> 
> sudo apt-get update
> 
> install wine Development branch
> https://wiki.winehq.org/Ubuntu
> 
> 
> open terminal and write(change USER for your user linux folder
> rm -fr ~/.wine export WINEARCH=win32 export WINEPREFIX="/home/USER/.wine" winecfg
> 
> select windows 7 emulation.
> 
> 
> create folder for winetricks(example mkdir winetricks)
> download winetricks last version on here type this
> wget -r -c -N https://raw.githubusercontent.com/Winetricks/winetricks/master/src/winetricks
> 
> cd /Winetricks/winetricks/master/src/
> sudo chmod +x winetricks
> 
> launch winetricks
> 
> navigate to the default wineprefix -> Installation of Windows DLL or component, find dotnet45 and mark it for installation. 4. Installer will install all .NET Frameworks including 4.5 version
> 
> system is fine for t2laser
> 
> now on terminal type winefile
> 
> search setup.exe (t2laser install app)
> 
> and enjoy.

Maybe someone wanna try the procedure with LaserGRBL?

Jakeler, 2018-01-17:
I have already tried it with Wine. dotnet does not properly install with latest winetricks (20171222) and wine (3.0rc6). You can use the open source wine-mono again, but then LaserGRBL does not even start, showing some NotImplemented errors...  
Usually the "native" mono is the way to go for .NET on Linux, i don’t have much good experience with wine.

closed 2018-02-27 by arkypita

MiguelAngelLV, 2018-11-14:
Any news about LaserGRBL in Linux?

Jakeler, 2018-11-18:
I don't think it is going to happen. 
If you are still searching for a solution, i am using [cncjs](https://github.com/cncjs/cncjs) now, which works nicely on Linux with GRBL. It provides a server and webinterface for machine control and preview. It has no path generation feature to import image/vector graphics though, so you need to do it manually, i use Inkscape with the J Tech Photonics Laser plugin to do this.

Langlais115, 2019-01-21:
I'm a little bit sad to here that. But I understand.
@arkypita 
Thank you for all your hard work.
Keep it that way ;-)

heltonritter, 2020-01-17:
Hey guys, ahhh are those changes related to run on Linux/Mono already merged to master right?

I read some other issues and seems that still some work to be done. I'm willing to take a look and see where I can help.

DanielMazurkiewicz, 2021-01-23:
> Hey guys, ahhh are those changes related to run on Linux/Mono already merged to master right?
> 
> I read some other issues and seems that still some work to be done. I'm willing to take a look and see where I can help.

Nope, just checked it, they are not

bcabral, 2021-06-30:
Hi there. I am facing the same issues and could not find https://github.com/arkypita/LaserGRBL/tree/Linux
Is there a place where I can download this compiled mono-capable-version of lasergrbl ?
Thanks

arkypita, 2021-07-01:
> Hi there. I am facing the same issues after download the latest version and tried on wine-5 and mono-6.12. Also could not find https://github.com/arkypita/LaserGRBL/tree/Linux

This branch was deleted because give no working results.

> Is there a place where I can download this compiled mono-capable-version of lasergrbl ?

No, this version doesn't exist.

It was reported by some user that LaserGRBL executable run good on Linux with wine 5.0 and wine mono 4.9.4 (wineprefix 32-bit) and windowsdll gdiplus.

[image]

bcabral, 2021-07-04:
Hello. Thanks for answering

I was able to install and run on my linux using the following procedure

$ sudo apt purge wine mono*
$ sudo snap install wine-platform-5-stable
$ export PATH=$PATH:/snap/wine-platform-5-stable/current/opt/wine-stable/bin/
$ rm -rf ~/.wine
$ export WINEARCH=win32
$ winecfg (choose windows 7 and exit)
download latest winetricks from https://raw.githubusercontent.com/Winetricks/winetricks/master/src/winetricks
$ chmod a+x winetricks
$ ./winetricks gdiplus
to free space after install gdiplus you can $ rm -rf ~/.cache/winetricks/win7sp1
download mono from http://dl.winehq.org/wine/wine-mono/4.9.4/wine-mono-4.9.4.msi and install with:
$ wine   msiexec /i  wine-mono-4.9.4.msi
download latest LaserGRBL (tested with https://github.com/arkypita/LaserGRBL/releases/download/v4.4.1/install.exe) and install with:
$ wine install.exe
$ wine "C:Program Files/LaserGRBL/laserGRBL.exe"

[image]

arkypita, 2021-07-04:
Thanks a lot, I believe that could be useful for other linux users. I will link your procedure on https://lasergrbl.com/faq

bcabral, 2021-07-04:
I'm glad I can help and thank YOU for releasing lasergrbl!

Cheers
!3runo
from Brazil

dandifon, 2021-08-08:
Hi. I'm a newbie in these matters. I have not been able to run laserGRBL using wine on linux peppermint. Wine development, winetricks, etc  are already installed. When I try to run laserGRBL, I keep getting the message that wine mono is not installed. I have followed several procedures on the Internet to get around this to no avail. Before trying the above steps which look wonderful, I just got an idea or question. Is it not a good idea to compile a stand-alone version of laserGRBL? I suspect it might run better on wine.

Langlais115, 2021-08-08:
> Hi. I'm a newbie in these matters. I have not been able to run laserGRBL using wine on linux peppermint. Wine development, winetricks, etc are already installed. When I try to run laserGRBL, I keep [...]

If you follow the installation procedure which is 2 post before yours, then try to start laserGRBL using **sudo**.
`sudo wine "/home/${USER}/.wine/drive_c/Program Files (x86)/LaserGRBL/LaserGRBL.exe"`

Although this is not a perfect solution, this might solve some permissions issue.

bcabral, 2021-08-08:
Hello

I don't use sudo. Just add yourself to dialout group and reboot (or exit and re-enter session) to be able to access the arduino device.

Try install mono as per my instructions, it should work

This said, the procedure I post is for ubuntu. I tried it in debian 9 and snap lacks some libraries that debian 9 don't have and won't install. There is an official winehq apt repository for wine5 that works for debian. I wrote a new procedure using it instead of snap and will post soon

In about a month using lasergrbl this way, I can say that works very well. If you pause and doesn't resume soon, the program will crash, but other than that I am very satisfied.

Cheers
!3runo
from Brazil

dandifon, 2021-08-09:
The procedure of bcabral works for me up to
$ export WINEARCH=win32.
winecfg returns 'No such file or directory'

bcabral, 2021-08-09:
You have to add wine binaries to your PATH. Where they are depends on how you install it

Cheers
!3runo
from Brazil

dandifon, 2021-08-10:
Thanks !3runo for being patient with me. winecfg works fine now. 
This link 'https://raw.githubusercontent.com/Winetricks/winetricks/master/src/winetricks' however gives but the source code of winetricks. I'm stuck there.
If I do 'sudo apt install winetricks' what gets installed is winetricks 20180207. I suppose that's old as the next command 'chmod a+x winetricks' yields 'No such file or directory' even though winetricks runs well on the same command line.
Please, what's the way forward?
Dan.

bcabral, 2021-08-10:
Download the link (or save as winetricks), chmod a+x and you're set!

i.e.

$ wget https://raw.githubusercontent.com/Winetricks/winetricks/master/src/winetricks
$ chmod a+x winetricks
$ ./winetricks

!3runo

dandifon, 2021-08-10:
WONDERFUL!  !3runo, THANKS A THOUSAND. laserGRBL now runs on Linux Peppermint on my computer.
Hahahahahahahahaha. It's exciting!
Thanks man. 
To arkypita, !3runo and Langlais115, I say  THANK YOU GENTLEMEN. 
Dan

ptorrezao, 2021-11-14:
> Hello. Thanks for answering
> 
> I was able to install and run on my linux using the following procedure
> 
> $ sudo apt purge wine\* mono\* $ sudo snap install wine-platform-5-stable $ sudo apt install libmpg123-0:i386 $ export PATH=$PATH:/snap/wine-platform-5-stable/current/opt/wine-stable/bin/ $ rm -rf ~/.wine $ export WINEARCH=win32 $ winecfg (choose windows 7 and exit)
> 
> Download latest winetricks from https://raw.githubusercontent.com/Winetricks/winetricks/master/src/winetricks $ chmod a+x winetricks $ ./winetricks gdiplus msxml3 to free space after install gdiplus you can $ rm -rf ~/.cache/winetricks/win7sp1 Download mono from http://dl.winehq.org/wine/wine-mono/4.9.4/wine-mono-4.9.4.msi and install with: $ wine msiexec /i wine-mono-4.9.4.msi
> 
> Add a string entry under HKEY_LOCAL_MACHINE\Software\Wine\Ports with a key of COM10 and a value of /dev/ttyUSB0 (or /dev/ttyACM0 if it is the case) to access the arduino USB port from wine and then create the symlink for it. The former is saved in .wine/system.reg $ wine regedit $ rm ~/.wine/dosdevices/com10 $ ln -s /dev/ttyUSB0 ~/.wine/dosdevices/com10
> 
> Add yourself to dialout group so you can access the USB port (if you're not already) $ sudo usermod -a -G dialout $USER
> 
> Download latest LaserGRBL (tested with https://github.com/arkypita/LaserGRBL/releases/download/v4.4.1/install.exe) and install with: $ wine install.exe $ wine "C:Program Files/LaserGRBL/laserGRBL.exe"
> 
> [image]

Thanks for the effort!! Working on 4.6.2!!!

arkypita, 2021-11-14:
Another penguin using lasergrbl, I'm happy!

Rybec, 2021-11-15:
I guess I should report my success with Play on Linux: I've replaced the fried motor controller, and it works perfectly fine now.  I do sometimes get some serious lag issues when running LaserGRBL through Play on Linux, but I think it is a Wine issue, not an issue with LaserGRBL.

bcabral, 2021-11-15:
Hello

As promised, an updated procedure without snap for debian

$ sudo apt purge wine\* mono\*

$ dpkg --add-architecture i386
$ wget -qO - https://dl.winehq.org/wine-builds/winehq.key | sudo apt-key add -
$ sudo apt-add-repository https://dl.winehq.org/wine-builds/debian/
$ sudo apt-get update
$ sudo apt-get install --install-recommends winehq-stable
$ wine --version
check if it is Wine-5

$ sudo apt install cabextract libmpg123-0:i386
$ export PATH=$PATH:/opt/wine-stable/bin
$ rm -rf ~/.wine
$ export WINEARCH=win32
$ winecfg (choose windows 7 and exit)

Download latest winetricks from https://raw.githubusercontent.com/Winetricks/winetricks/master/src/winetricks
$ chmod a+x winetricks
$ ./winetricks gdiplus msxml3
to free space after install gdiplus you can $ rm -rf ~/.cache/winetricks/win7sp1
Download mono from http://dl.winehq.org/wine/wine-mono/4.9.4/wine-mono-4.9.4.msi and install with:
$ wine msiexec /i wine-mono-4.9.4.msi

In debian the /dev/ttyACM0 arduino port is automatically added as com5

Add yourself to dialout group so you can access the USB port (if you're not already)
$ sudo usermod -a -G dialout $USER

Download latest LaserGRBL (tested with https://github.com/arkypita/LaserGRBL/releases/download/v4.6.0/install.exe) and install with:
$ wine install.exe
$ wine "C:Program Files/LaserGRBL/laserGRBL.exe"

Hope this helps
Cheers
!3runo

oliverhbailey, 2021-11-18:
The members in this thread have really gone the distance in getting LaserGRBL to work on Debian. I worked on the original NT project back in the early 1990's. As an operating system developer for almost 50 years, I still develop real time systems for several industries, using Linux almost exclusively these days.
Latency can be caused by several factors. LaserGRBL is a 32 bit windows application. Under Wine, the Dll's have access to half as much memory than under a Windows environment.  A 32 bit application normally has access to 4 gigs of RAM. But under Wine 32, only the lower 2 gigabytes is actually available. There are other factors as well,  type of machine, type of processor, speed, cores, and total ram, that also have an impact on performance of a Windows app.
The best solution is to move LaserGRBL to 64 bit Wine and mono. That eliminates memory and CPU restrictions of Wine 32 and should done anyway as 32 bit Windows support is deprecated. I'm retired and ported LaserGRBL to 64 bit Mono swveral yrars ago successfully. But in the past two years Microsoft has stopped promising cross platform support for Winforms on Linux only. And at that time and continuing to the present, there are build errors that remain unresolved on Linux relating to Winform 32 bit only applications. 
There are ways to bump up the application process priority but this can cause other performance issues. If the latency is in the UI, its most likely memory starved. However, if it is in data buffering to the laser, you could increase the buffer size to keep data flowing smoothly. Just don't increase the buffer until you have other performance problems.
Wine is useful, but trying to solve Windows shortcomings is getting to be a bigger burden. Memory sensistivity is even caused by the order of Dll loading these days on 32 bit applications.
You've done a great job of getting LaserGRBL working. I hope this information can help.

Styne13, 2021-11-20:
@Rybec: Thanks for describing installation using Playonlinux. I created a POL-Installation-Script to automate most of the steps including download of latest version and com port mapping in regedit.

@arkypita: Shall a create a PR to add it to the repo? What is the best path to put it in?

Rybec, 2021-11-20:
Thank you so much!  I was hoping someone could use my process to do this.

Styne13, 2021-12-02:
I created a pull request with a setup script now:
#1666
Feel free to comment!

arkypita, 2021-12-08:
Hi @Styne13 
I see that there is something similar to PlayOnLinux but for MAC https://www.playonmac.com/en/
Do you think it is possible to do something similar to your script but for MAC?

Styne13, 2021-12-11:
Hi @arkypita 
think that shall also be possible. As far as I know the interface between wine and playonlinux and playonmac is identical. Executing installation scripts is also possible.

It might depend on the version 5.22 with mono available in macOS.
Gave it a quick try with my 1201n hackintosh running Yosemite was not successful, because I could not install the appropriate version 5.22 with mono.
Maybe someone else running a newer version of macOS can give it a try.
Can support as possible and also give it another try installing mono manually...

arkypita, 2022-02-05:
@Styne13 I have installed a virtual box with macOS Big Sur. Can you give me some hint on how to use playon### tools and how you build your PlayOnLinux script? I would like to try doing the same for mac.

arkypita, 2022-02-05:
> First I installed the latest version of Wine 5 in PlayOnLinux (32 bit version, it was 5.22, not the staging version), under Tools > Manage Wine Versions.

Sh*t, I am already stuck on first step! I cannot find "Tools > Manage Wine Versions" in PlayOnMac UI.
Nothing under "Impostazioni" (settings) an nothing under "Configura" (configure). Any Hint?

[image]