arkypita/LaserGRBL #6: Feed override

jiarko, 2017-01-04:
Hi,

I'm testing your GUI in "simulation" (I mean I don't have the real CNC laser on, just the controller) and playing with feed override sliders I see no changing in the mainwindonw status bar neither any speed changin into the cursor moving while streaming the GCode.

can you help me on this?

btw, very good software!!! 

P.S. what is the power of your laser CNC?

arkypita, 2017-01-04:
First, the feed override feature was added in grbl 1.1 and is not present on previous versions.
LaserGRBL detects the grbl version at connection, and it should show the slider controls only if the version is> = 1.1
I assume it for sure, but you can confirm that you're using a 1.1 version of grbl on the controller?

Grbl only has only an [increase/decrease interface](https://github.com/gnea/grbl/blob/master/doc/markdown/commands.md#extended-ascii-realtime-command-descriptions) for setting the override value, and give a feedback of the actual override by the real time status report message.

I have implemented the "slider" feature by reading the actual override in real time status report, and then sending increments/decrements until required override is reached.

In order to works certain conditions are necessary:
- you must not change CMD_FEED_OVR_*, CMD_RAPID_OVR_*, and CMD_SPINDLE_OVR_* in config.h
- you must not change MIN/MAX FEED_RATE_OVERRIDE, COARSE_INCREMENT, FINE_INCREMENT etc in config.h
- you must not have disabled #define REPORT_FIELD_OVERRIDES in config.h

Of course there is also the possibility that I have done something wrong in implementing reading the status report message, and then not be able to read the current values override.

arkypita, 2017-01-04:
To allow me to better understand what could be the cause, I ask you to:
- Tell me the exact version of grbl in use, obtainable with the command $I
- Attach your config.h
- Attach your grbl configuration (exportable with LaserGRBL: Menu-Grbl-Export Config)

jiarko, 2017-01-04:
Hi,

I use grbl 1.1e, 

[my_config.txt](https://github.com/arkypita/LaserGRBL/files/686019/my_config.txt)

basically I've loaded the HEX file from grbl github so I dont use any config.h.. (if I properly get your question).

just for you to know, I'm using only the arduino uno flashed with the grbl 1.1e for "testing purpose" I don't have a complete cnc build upon it now with me.

P.S. where do you come fom? (I'm from Italy too)

arkypita, 2017-01-05:
the problem is with $10=1, try change to $10=19 and it will work
when I have a little of 'free time I will analyze the problem, thanks for reporting

(i am from Bergamo)

arkypita, 2017-01-05:
Ok... my code suppose that Ov (override reports) is in position 4 of a splitted string, splitted by | character:

`string[] arr = rline.Split("|".ToCharArray());
if (arr.Length > 4 && arr[4].StartsWith("Ov"))
ParseOverrides(arr[4]);`

This work with $10=19 that produce this output

Idle|MPos:31.795,1.000,0.000|Bf:15,128|FS:0,0
Idle|MPos:31.795,1.000,0.000|Bf:15,128|FS:0,0|WCO:0.000,0.000,0.000
Idle|MPos:31.795,1.000,0.000|Bf:15,128|FS:0,0|Ov:100,100,100

But not with $10=1 that output a shorter data string

Idle|MPos:31.795,1.000,0.000|FS:0,0
Idle|MPos:31.795,1.000,0.000|FS:0,0|WCO:0.000,0.000,0.000
Idle|MPos:31.795,1.000,0.000|FS:0,0|Ov:100,100,100

arkypita, 2017-01-05:
This version fix: https://github.com/arkypita/LaserGRBL/releases/tag/v1.2.2

closed 2017-01-05 by arkypita