arkypita/LaserGRBL #10: Development Roadmap

arkypita, 2017-02-01:
@ January 15, 2017, I had to suspend the development of LaserGRBL because currently I have several other work projects. The development will resume when I have some free time to devote to this project.
The support for any bugs and issues reported via Github will remain active.

Planned developments for LaserGRBL are:

- Picture framing, helping to position the workpiece. (Laser go around the pictures at the lower power)
- DXF import

Further requests for new features can be added at the bottom of this post.

wfluna, 2017-02-15:
Greetings. It is a very interesting project. I am test your software. You must rename project with other word ex. LaserGrbl + ...... It is difficult to find, the name is very generic. and You can put speed control near to Joggin panel. 
See a rule in Import raster image  for approximate measurements
Congratulations
Good Look 
(Buena Suerte)

arkypita, 2017-02-15:
Hello Fernando
Thank you for your appreciation, and suggestions.

I'm not sure I understand what you mean with: "See a rule in Import raster image for approximate measurements". You can try to explain it in another way?

wfluna, 2017-02-16:
Ok
Sorry. My native language is Spanish.
When you open a new file.
Open “Import raster image” dialog box
In preview view is useful a rule in x, y coordinates to appreciate image measures.

Best Regards
Fernando

s

kefeur, 2017-02-19:
Hello Master.
Request proposals I turn to you
My lasers are low power (250mW, 70% of this output has not been burnt).
1; It would be advantageous to convert the dirthering filter photos, burn the wood using a lot of points. 1 bit, there are points - no points ....
2; It should be used in the Z-axis of the laser focus. But at least it should be on the up and down buttons. This could be either automatic probe as well.
3; If this is difficult, it could be a user-defined buttons.
Example:
Button 1: G10L2 (P1) X0.0 Y0.0 // current image of 0
Button 2: Start G21G91G38.2Z-30F100 // Z probe.
Etc...
The user can give the function of each of the four buttons.
Thank you for your attention.
Sorry for my bad english.
Ferenc

arkypita, 2017-02-20:
About @wfluna request:

> Open “Import raster image” dialog box, In preview view is useful a rule in x, y coordinates to appreciate image measures.

Raster image does not have a "real size" if not in pixel.
Conversion between pixel size and printed size could be done only using an arbitrary dpi resolution: [about image size](https://www.creativeprogression.com/resolution/)

More so, a vector image does not have a "real size", cause it can be scaled to any size without loosing definition.

In the second form of the import process, the one opening when you press "next", you can choose the size of your final image (and engraving offset) independently from the original image size in px.
All the conversions, upscaling and downsampling of the image are managed internal by LaserGRBL to produce the best result for your engraving resolution (line/mm) and final size.

So, why you want to define the size of the imported image?

arkypita, 2017-02-20:
About @kefeur requests:

> My lasers are low power (250mW, 70% of this output has not been burnt).
> 1; It would be advantageous to convert the dithering filter photos, burn the wood using a lot of points. 1 bit, there are points - no points ....

I know your setup, my first laser was 250mW red and it did not burn anything other then leather (now i have 2W blue). You talk about wood? It sound very difficult to me.

About dithering suggestion I am not sure to be skilled enough to implement it, but before that I doubt it will work really. If I'm not mistaken, dithering would be use the laser always at full power, doing more or less dense dots to emulate grayscale.

This is nothing more nor less than what makes the PWM modulation itself, so if you have trouble with power modulation where at 70% of this output has not been burnt, you will have the same problem with dithering where image is brighter.

If you're good enough, you can try yourself to make a test manually generating GCode (eg using Excel) and producing a scale of tones with dithering technique.

> 2; It should be used in the Z-axis of the laser focus. But at least it should be on the up and down buttons. This could be either automatic probe as well. 3; If this is difficult, it could be a user-defined 
> The user can give the function of each of the four buttons.

I like user defined buttons, it could be a nice new future feature.
I added it to development roadmap.

> Sorry for my bad english.

My english is bad too, hope you understand

kefeur, 2017-02-20:
Hello Arkypita
Thanks for the reply.
I would like to congratulate this very good software.A great help to me.

I tested the difference between the PWM  conversion, and the dithering conversion.Both can be very good . Created 15 pieces of image based on photo.
[image]
This is the result, I think this is a better solution. Because the laser to burn 100% capacity, and each point is very clearly visible.I have the right dithering technique ,for my low power lasers .
Thank jou 
Ferenc

arkypita, 2017-02-20:
Hi Ferenc
I congratulate you on your job, this is a very good result. Even with my most powerful lasers have never been able to get a result as yours with PWM tecnique cause of a very small range of reproducible shades 

In my setup PWM from 0 to 30% result in totally white, 30 to 60% light burn with very small difference, 60 to 100% strong burn with very small difference).

I am very interested to include dithering feature in LaserGRBL, if this could be the result. Can you tell me what steps you used to get your picture?

wfluna, 2017-02-21:
About @arkypita  requests:
So, why you want to define and view the size of the imported image?
I understand image and size position is useful 
[image]
but is useful too see image measurement  in preview view because we need cut with cut with and
specific Long and wide .
sometimes picture is small and other times very big in relation with work area.
[image]

and in ## vectorize menu is necessary some similar to Inside first. to cut first inside of letter ex.
A. it is necessary cut inside first. other way letter drop out.

I hope to open in vs2015. I will write about it.
Best Regards

kefeur, 2017-02-21:
Hy Arkypita
I use software : IrfanWiew with plugins of XiGrayDither_sp 

other facilities:
[http://www.tannerhelland.com/4660/dithering-eleven-algorithms-source-code/](url)
[https://github.com/villamany/3dpBurner-Image2Gcode](url)

I am so I use IrfanView:(It is the best for this purpose)
Download IrfanWiew , download plugins of XigrayDither
1, start IefanWiew , add the plugins
2, open the image
3 click image button , click Adobe 8BF plugins , click  XiGrayDither_sp.8BF
4 select RGB , select levels =2, select Floyd-Steinberg filter ,
5 calcute prewiew
.....this completes the two-bit image.
Open the LaserGRBL -line to line conversion - 10 lines/mm ....feed rate 300to600..

Original image:
[image]
Converted image:

[image]

This type of wood has not created image
Burned image of wood:(Wrong photo)
[image]
LaserGRBL transformed the personal use is not finished yet, there is no time for it. I hope you do not mind.Sorry.
[image]

arkypita, 2017-02-21:
Hello @kefeur 
Driven by your results I try to use dithering to mark an image.

Original:
[image]

I have converted the image with photoshop:
[image]

Then engraved it with LaserGRBL with NearestNeighbour scaling (do not blur if scale), 8 line/mm.
[image]

This is the result:
[image]

As you suggest, the results are better than the PWM marking.

I think i will include dithering inside LaserGRBL as soon as possible, cause i like it.
Alghoritm can be found on the web: [https://www.cyotek.com/blog/dithering-an-image-using-the-floyd-steinberg-algorithm-in-csharp](https://www.cyotek.com/blog/dithering-an-image-using-the-floyd-steinberg-algorithm-in-csharp)

> LaserGRBL transformed the personal use is not finished yet, there is no time for it. I hope you do not mind.Sorry.If there is a problem with this, please let me know, delete or edit or....

It's ok to do that, enjoy!

kefeur, 2017-02-22:
Hello Arkypita.
Congratulation.
How to use the converted image of the LserGRBL?
What were the options?

arkypita, 2017-02-22:
It was only a try, image converted with Photoshop. Ditering not implemented yet in LaserGRBL, will be implemented when I have spare time.

arkypita, 2017-02-25:
@kefeur [Dithering ready](https://github.com/arkypita/LaserGRBL/releases)!

arkypita, 2017-02-25:
> and in ## vectorize menu is necessary some similar to Inside first. to cut first inside of letter ex.
> A. it is necessary cut inside first. other way letter drop out.

@wfluna I understand that it could be very usefull for you, but this is very very difficoult to implement.
Yes, in your "A" letter example could be easy, but in a more complex drawing it could be tricky to detect what is inside and what outside.

However, there is a very simple program called "inkscape" that can do vector graphics. Installing "[Jteck photonic laser tool](https://jtechphotonics.com/?page_id=2012)" plugin it turn in a very powerfull GCode generator. You can draw your letters in inkscape, transform then to path, separate internal and external, then generate one GCode file for internal, and one for external, and so cut one than the second using LaserGRBL or others tool.

This could give you more control on the final work, i suggest you this way.

wfluna, 2017-02-26:
Tks a lot, @arkypita .

kefeur, 2017-02-27:
@arkypita  
I am very happy .
Thanks.

arkypita, 2017-02-27:
I'm working on custom buttons, stay tuned

kefeur, 2017-02-27:
Very good

kefeur, 2017-02-27:
We have a problem: no engraving speed box
[image]
 :

arkypita, 2017-02-27:
Thanks for feedback, It was an oversight.

For the unnecessary dotted lines, I saw them too. They are generated by the algorithm of dithering itself, and I have no knowledge of how work. Does they disappear if you play with contrast/brightness?

New version released.
Fix missing speed control & add custom bottons (experimental). Try to right click in lower buttons area right to the unlolock button to add a custom one.

kefeur, 2017-02-27:
Speed controll is ok.

kefeur, 2017-02-27:
Next test not ok. User definied buttons program break..
Decimal values are too large or too small..

arkypita, 2017-02-27:
I developed the functionality without being able to test because I do not have available laser at this time. I will do some test later

kefeur, 2017-02-27:
Ok . Thank you.

arkypita, 2017-02-27:
I have added a lot of features in the last week, driven by some user suggestions. Now it's time to stop and wait for user feedback and bugreport.

My #1 priority now is to provide some kind of documentation/wiki for the project.

kefeur, 2017-02-27:
I mean, it would be a few more comments, the g-code optimization software.
Please let me know when it is appropriate to discuss.
You may mail in private if it is good.

kefeur, 2017-02-27:
What is the part of generated G-code?

kefeur, 2017-02-27:
When it starts the burning the M3 instruction is omitted in the first line. At the end of the lines it's an M5 then when it starts coming back M3, this is unnecessary. engine driver on off on off)

arkypita, 2017-02-27:
To keep a a proper categorization of topics, I invite you to open a new issue for each different question.
No bug report here, only feture request please.

Feel free to open new issue as you need

kefeur, 2017-02-27:
Good job.
Thank you.

arkypita, 2017-03-20:
Documentation done, currently working on Linux/Raspberry port

arkypita, 2017-03-22:
The development of a version for raspberry proceeds with difficulty. The version of Mono distributed with raspberry is very old (3.2.8) and suffers from several bugs that prevent winform program to function properly. I'm trying to update the mono version by following some guides I found on the internet, but with no success.

arkypita, 2017-04-03:
Added experimental support for PWM-less hardware: https://github.com/arkypita/LaserGRBL/releases/tag/v2.5.1-pre

arkypita, 2017-04-08:
Added dithering algorithm selection

alfredanil, 2017-06-01:
Do you plans for 3D laser engraving, more specifically I was looking for acrylic bending like this https://www.youtube.com/watch?v=arjRtCjI9AQ
Your doing a great job, I really admire your work. I have not a laser engraver yet but in process of developing one with practically no belt elongation/streching at high speeds (+70mm/sec ) and no it is not a linear motor.

mkeyno, 2017-06-06:
@arkypita  may I suggest to add the wifi serial bridge to your milestone ? I would love to be helpful  in this step

arkypita, 2017-06-07:
@mkeyno Yeah! It could be a very nice feature in LaserGRBL.

Unlucky i cannot test it easly, since my hardware doesn't have WiFi (i have some WiFi modules in my lab, but i never try to use-connect them).

I think that on LaserGRBL side i only have to implement a tcpsocket to use to connect to a specific IP address that stay in listening on a specific ip:port.

Is this correct?

mkeyno, 2017-06-07:
yes it is so easy just like the Bluetooth hardware connected to ESP8266 RxTx and server websocket listening to any request came by LaserGRBL , ESP8266 server websocket is already stable and use for many project such as [ESP3D](https://github.com/luc-github/ESP3D)  & [laserweb ](https://github.com/LaserWeb/LaserWeb3), you just need to add websocket client in your  program , I think in previous ticket I sent you its library

arkypita, 2017-06-10:
@mkeyno please switch to #51 thread

arkypita, 2017-07-14:
Close this roadmap, create a new one.

closed 2017-07-14 by arkypita