arkypita/LaserGRBL #12: Visual studio Community 2015

wfluna, 2017-02-20:
Hi.
I try to open this project in Visual Studio Community 2015 but 
many errors appear.
What is the best integrated development environment (IDE) for this project?

arkypita, 2017-02-20:
LaserGRBL is developed with SharpDevelop and is written in C# language targeting 2.0 framework, so it is fully compatible with older versions of Windows XP (which can also be compiled). I also tryed to open it with VS2012 and it work.

Anyway the code is all inside the solution (no external dll, no NuGet component, nothing) and must be possible to convert and open in any newer versions of the IDE, or target it to newer framework version like 3.5 and 4.0.

arkypita, 2017-02-20:
Try to checkout now, i have just committed a version with double .sln file: one for SharpDevelop and one for VS2012. Try open VS2012 version with VS 2015, it should convert and work.

If not, please tell me more about errors

wfluna, 2017-02-21:
hi @arkypita 
[image]

click in ignore and continue 
[image]
and in the preview form label
[image]
Best regards

arkypita, 2017-02-21:
Are you sure that you have compiled in the solution at least once time?
Please use "Build solution" from "Build" menu. It must compile, errors will disappear.
If you have compile errors, please send (copy-paste) the full build output ("View" - "Output")

wfluna, 2017-02-23:
no problem anymore, "Build solution" from "Build" menu is ok
and with this the app work

wfluna, 2017-02-23:
and with sharp development this work well too.
Thanks a lot.

closed 2017-02-23 by arkypita