arkypita/LaserGRBL #14: G code start problem , and g code optimisation.

kefeur, 2017-02-27:
When it starts the burning the M3 instruction is omitted in the first line.
At the end of each line M5 switches off, then when it starts coming back M3 switches back on. Consecutively, both of them are unnecessary.

kefeur, 2017-02-27:
G code Optimisation:
In an example of a 50mm*50mm picture that has a 20mm wide figure in the middle and has 15mm of empty space the laser travels that empty space as well but does nothing. It would be better if it would skip these empty spaces and empty lines entirely.

White skipped the purpose / work transcends time and unnecessary abbreviations Movement
It would be good if it would skip the white parts during the work process to shorten the time and prevent useless movement.
Thank you.

arkypita, 2017-02-28:
> When it starts the burning the M3 instruction is omitted in the first line.

If so, it's a bug. I will investigate as soon as possible.

> At the end of each line M5 switches off, then when it starts coming back M3 switches back on. Consecutively, both of them are unnecessary.

This is necessary and by design.

[image]

Take a look at this image, made line by line by horizontal strokes. The horizontal line must have the laser on, the vertical line exchange must ensure that the laser is switched off. Otherwise it would be marked a vertical line to the left and right edges of the image.

It's true that in cases where at the end of the line laser is already off (S0) this M5-M3 seems superfluous, but it is not true in all other cases.

arkypita, 2017-02-28:
> In an example of a 50mm*50mm picture that has a 20mm wide figure in the middle and has 15mm of empty space the laser travels that empty space as well but does nothing. It would be better if it would [...]

Yes, a lot of improvements can be made in this direction.
All leading / trailing white lines may be skipped. Also avoid arriving at the end of line but move directly to the next line. Something like this:

[image]

In the past I tried to insert optimizations such these, but the line generation algorithm became too complicated so i abandoned this way.

kefeur, 2017-02-28:
> It's true that in cases where at the end of the line laser is already off (S0) this M5-M3 seems superfluous, but it is not true in all other cases
>

Many other g code generator did not use the M3-M5 is anything but start and end. Use S0 instruction .The problem to quickly turn off and then turn the engine driver, and because the micro step is clapping, jumping and noisy. sometimes jumps one step  it is a lot . (1 step = 64 micro step)
Im Isolating the engine driver enable input from the GRBL.

kefeur, 2017-02-28:
M3 error.
 G code  first line:

G90
F400
G0 X0 Y0
M5 S255
G1 F350
G91
G0 F400 X2.6 S0
G1 F350 X.1 S255
G0 F400 X2.1 S0
G1 F350 X.1 S255
G0 F400 X.5 S0
G1 F350 X.1 S255
G0 F400 X2.3 S0

This is correct:

G90
F400
G0 X0 Y0
M5 S255
G1 F350
G91
M3<<<<<<It can not generate it
G0 F400 X2.6 S0
G1 F350 X.1 S255
G0 F400 X2.1 S0
G1 F350 X.1 S255
G0 F400 X.5 S0
G1 F350 X.1 S255
G0 F400 X2.3 S0

arkypita, 2017-03-01:
If you are able to checkout sourcecode and compile, the last commit has a fix for this bug.

arkypita, 2017-03-01:
GCode optimization was added in development roadmap (when I have time...)

closed 2017-03-01 by arkypita

kefeur, 2017-03-01:
Very clever man.
Thank you.

arkypita, 2017-03-02:
Last commit contains GCode optimization

arkypita, 2017-03-02:
About the problem you tell me about dithering, of sparse small dots 

[image]

It not depends on a bug in the dithering alghoritm but it depends by the source image. If the source image has not a perfect white background (ie. for jpg compression artefacts) dithering try to represent this non pure white areas/points with small scattered dots.

If the image has a true white background this problem disappear

[image]

Maybe I could add some kind of "white thresholds" to clean dirty background in image preprocessing, for now I suggest you to preprocess images with Photoshop or similar to clean background.

kefeur, 2017-03-02:
The G-code optimization has been very good.
This is a very nicely developing project.
Thank you.

commit 296048c4 (2017-03-01): Fix #14 issue