arkypita/LaserGRBL #15: language

wfluna, 2017-02-28:
Hi @arkypita I want to translate to Spanish . I translate in my test software
but i think that is a good idea have an option language to translate to other language
Best Regards
Fernando

arkypita, 2017-02-28:
Of course you can change the various pages of LaserGRBL (fortunately not many) and replace all English strings with their Spanish translation. This work, however effective, is not very functional.
Each new release require you to repeat the process and your work would not be shared with others who appreciate the Spanish version.

With Visual Studio it is possible to create a multilingual application. To do that certain guidelines must be followed: https://msdn.microsoft.com/it-it/library/1021kkz0(v=vs.100)

In practice, the visual studio project should be set up to support multi-language, the programmer sets what language want to edit, and under the hood visual studio builds different sets of strings. It sounds very simple, but I have never done this kind of work with VS, so I do not know well how to do it.

The best thing I think is that I experience this feature making the translation in Italian (which is my native tongue) and then I will ask you to make a Spanish version.

arkypita, 2017-03-02:
I have experienced enough with the translations, and I created the Italian version. It was not a particularly difficult job.

From what I see, this is the first time you use Visual Studio. If you want to take care of the Spanish translation of LaserGRBL I can explain how to proceed.

wfluna, 2017-03-03:
tks. I am interesting in translate to Spanish and maintenance lasergrbl in spanish.
please explain me and I make it.

wfluna, 2017-03-03:
I downloaded last version and have 2 folder (English and Italian ) where i put or choice (.dll)to test in Italian

arkypita, 2017-03-03:
Windows select automatically the best available language (between available ones) based on language locale settled in windows itself. This means that on an Italian PC it will select ita translation, on a English (uk, usa etc) it select English.

So, once implemented spanish translation the selection of this language will be full automatic.

To test italian translation you need to change your system language.

Give me some minutes and i will write you how to make spanish translation

arkypita, 2017-03-03:
In visual studio, open the form you want to translate. In form properties there is a "Language" property.

[image]

Select "Spagnolo" (I think is "Espagnol" in you version) 

[image]

Now you can edit all the text and tooltip property for each control. The translated strings will be saved in a special resource file called "formname.es.resx"

[image]

When compiled the output folder will contains a localized dll resource file with spanish, and LaserGRBL now run in spanish on spanish environement.

arkypita, 2017-03-03:
To share with me your translation, please send all the .es.resx files uploading them in this thread.

arkypita, 2017-03-03:
Note: not all the string in software could be translated this way, cause there are some strings that are dynamically generated in code or concern a particular set of enum values (like grayscale algoritms).

This is not a big problem, string in form are 99% of all the strings, other strings can remain in english.

wfluna, 2017-03-03:
ok. these days I translate and send you.

wfluna, 2017-03-03:
This is folder
[.es.resx.zip](https://github.com/arkypita/LaserGRBL/files/817779/default.es.resx.zip)
with es.resx files , translations is Spanish.

wfluna, 2017-03-03:
I test it in a windows xp machine (sp2) and run in Spanish (VS compiled), some missing text, but with time. I review and solve it.
tks. Its is great. (This is my first software translation).

closed 2017-03-05 by arkypita