arkypita/LaserGRBL #19: Is there a setup tutorial for the Linksprite Laser head?

rseijo, 2017-03-12:
Hi, I have the Linksprite CNC with the optional Laser (pls see below).  
Does anyone know how to properly setup, adjust and or focus the Laser for Etching vs Cutting?   
I have tried to locate instructions for this laser with no success. 

I will be grateful for any guidance on this...

Thanks!
Ralph

http://linksprite.com/wiki/index.php5?title=DIY_CNC_3_Axis_Engraver_Machine_PCB_Milling_Wood_Carving_Router_Kit_Arduino_Grbl
 
Laser Head for DIY CNC 3 Axis Engraver Machine:  **KIT_CNC_LASHA**
Laser Wavelength 405nm
12V DC  
Output Power  500mw.

http://linksprite.com/wiki/index.php5?title=Laser_Head_for_DIY_CNC_3_Axis_Engraver_Machine

[image]

arkypita, 2017-03-12:
A perfect laser focusing is necessary for cut and engrave too, specially 
with low power laser.

This is what I do when I nedd to focus my laser:
- Put a piece of black cardboard on the engraving surface.
- Turn on laser at the lower available power M3 S30
- Do a small move G1 X1 F100
- Now focus by hand, rotating collimation gear to obtain the smallest dot.

Try this sequence
Diego

Inviato con AquaMail per Android
http://www.aqua-mail.com

arkypita, 2017-03-12:
Ps: always wear safety glasses when near look to focus point!

Inviato con AquaMail per Android
http://www.aqua-mail.com

rseijo, 2017-03-12:
Arkypita, thank you so much!   I will focus the laser as you recommend.  And yes, laser safety glasses!

Thanks again
Ralph

rseijo, 2017-03-12:
Diego,  Being that the Linksprite CNC has a movable Z axis, is there a specific height the that laser should be at for focusing? 

Thanks

arkypita, 2017-03-12:
I dont' know, I have a different laser and my CNC is 2 axis only, no Z.

closed 2017-03-13 by arkypita