arkypita/LaserGRBL #23: Laser ON/OFF (M3/M5) dont work?

Ch3fk0ch88, 2017-03-25:
Hi, i trie to engrave something like this:
[image]
But the result looks like this:
[image]
I testet the command m3 and m5 and the laser goes on and off.

arkypita, 2017-03-26:
LaserGRBL use "S command" to reproduce pure black or white image as well as greyscale image.
"S command" control power modulation: the ability of arduino to generate a PWM signal that control the instant power of laser by switching it very fast through its driver "TTL" pin (M3/M5 could not switch laser so fast).

Indeed, to engrave an image composed only by the colors black and white would suffice the M3 and M5 commands, but to take full advantage of laser potential, the PWM modulation is really important.

arkypita, 2017-03-26:
Please look at those link to read about power modulation and how to activate in your board.
http://blog.protoneer.co.nz/grbls-pwm-functionality-how-to-control-a-laser-or-variable-speed-spindle/
https://github.com/gnea/grbl/wiki/Connecting-Grbl#grbls-pins (PWM use pin 11)

How to modify driver to support laser power modulation:
https://forum.arduino.cc/index.php?topic=192733.0
https://forum.allaboutcircuits.com/threads/ttl-modulated-driver-for-laser.31275/
http://forums.reprap.org/read.php?133,703931

In addition version 1.1 grbl has added an important new feature for those who use the laser, which is precisely the "laser mode". To read anything more informed about it here:
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode#laser-mode-overview

If you follow my instructions and you will activate the laser power modulation you will find a tool much more useful than what you have now.

Best regards

Ch3fk0ch88, 2017-03-26:
Hi,
thanks. 
at the moment i only can turn the laser on and off (have to reasearch on which way i can add a pwm inkl so DDL)

arkypita, 2017-03-26:
Maybe I could add a config switch for who does not have PWM ability that replace all S command with M3/M5 command based on threshold. I added it on dev roadmap.

Ch3fk0ch88, 2017-03-26:
Ok, thanks.

Ch3fk0ch88, 2017-03-26:
So,
i tried to use an PWM like this:
[image]
When i laser something it looks like it works (a bit).
But:
1. I dont get the full power
2. I cant enable the laser, only when it "draw something" it goes on. I cant turn it on manually to focus the laser for example.

(Just for Info)

arkypita, 2017-03-27:
N° 1
You can not get full power even if you connect the control pin to fixed +5V?
S255 must produce a PWM modulation of 100% time +5V out of pin 11, so it is equivalent to a +5V when you issue a S255 command.

I suppose you don't have an oscilloscope to test signal on D11 but you could try to do some diagnoses with a led+resistor on D11, and with a multimeter on transistor Vce to see if you loose some power on transistor.

Also remove 10uF capacitor near Laser Diode. PWM control need to be fast in on/off operation, and capacitor add charging/discharging times when transistor open/close to ground.

P.s. i usually prefer MOSFET as switch when i design PWM circuit, but transistor could be ok too.

N° 2
https://github.com/gnea/grbl/issues/72
https://github.com/arkypita/LaserGRBL/issues/19

Ch3fk0ch88, 2017-03-27:
Hi, no i dont have a oscilloscope.
When i connect the E3055 to D12 and disable #define VARIABLE_SPINDLE than i get full power.

I anyway want to test a MOSFET or a 555 timer as next step, greetings.

arkypita, 2017-04-02:
Hi! Any news? Did you get it working with full power?

Ch3fk0ch88, 2017-04-02:
Hi, 
I have not ordered the things yet.
I wait to make a bigger order cause of the shipping costs (shipping is more expensive than the part itself^^)

arkypita, 2017-04-03:
I know, it's a common problem. 

By the way I don't know if LM317 could be easly used in a TTL circuit.
http://laserpointerforums.com/f67/pwm-ttl-controlled-driver-56027.html

Maybe it's easier to buy a ready laser driver with ttl like this one: http://www.ebay.co.uk/itm/12V-TTL-200mW-To-2W-445nm-450nm-Laser-Diode-LD-Power-Supply-Driver-Board-/322453276000?hash=item4b13b68560:g:a2UAAOSwhQhYyCzg

arkypita, 2017-04-03:
In these days I made a version that allows to use LaserGRBL on machines that do not have the PWM control. I have not had a chance to test it properly, but if you want to try it and tell me if it works you can download it from here: https://github.com/arkypita/LaserGRBL/releases/tag/v2.5.1-pre

Ch3fk0ch88, 2017-04-03:
I will test it net weekend.

Greetings.

Ch3fk0ch88, 2017-04-07:
Hi,
i testet the Version.
But why i can only use Vectoring and this 1bit thing when i disable the PWM Support?
With this i 1 Bit it all time looks a but broken.
[image]
[image]
[image]

Greetings.

arkypita, 2017-04-07:
The function called "Line To Line line tracing" is used to produce grayscale images modulating power, so it cannot be used for hardware setup without power modulation. So I have removed the option for PWM-less hardware.

1 bit dithering try to reproduce eye grayscales effect by creating a black points map. The dithering alghoritm could add some artifacts when white pixels are not "pure white".

http://lasergrbl.com/usage/raster-image-import/dithering-tool/
http://codegolf.stackexchange.com/questions/26554/dither-a-grayscale-image

But... your image is not a grayscale! Is a pure black logo. The best result for pure black logos is obtained with vectorization + filling. Try this combination, result will be OK!

closed 2017-04-18 by arkypita

joncarns, 2018-10-13:
I have a similar issue, I have been able to connect. I can move the steppers however laser stays on from start to finish. if its plugged in the computer it is on regardless of having lasergrbl started or not. ive adjusted $32=1. unchecked the pwm box. didnt have this issue until i uploaded grbl v1.1.hex but i had no connectivity before that. ive ran the laser with t2laser however subscription issues rendered my computer without a hardware number. ive tried S0 m5

StuartB4, 2018-10-13:
Hi joncarns. 
What setup do you have?
Which laser, laser driver, Arduino Type etc, 
Do you have any pics of the wiring / connections of the machine.

facumaster, 2019-04-29:
hello, im using versión 3.0.10 and i cant make laser work with M3
ive already checked pin D11 and it gives 5v with M3 command, but when i plug PWM laser's pin, it does nothing. 
movements work alright, very smootly
any ideas?

[image]

[image]

StuartB4, 2019-04-30:
@facumaster

[image]

facumaster, 2019-04-30:
yes i do, ive tried the pulldown resistor solution aswell (1k ohm), but no luck.

Emoslapper, 2019-10-28:
G'day All,

@StuartB4 I have read in other responses that you have a dvd\cd setup as do and I.
And I am running a Arduino Uno + CNC Shield V3 on the Z+ Stop for laser on, just using a 200-250mW 650nm Laser. So my question is:
Is it possible to run a PWM setup on my system?
I have the PWM ticketed on but when I use m4 the laser just does not turn on and on m3 it works.

Cheers

StuartB4, 2019-10-28:
It should work as it is providing you laser driver is PWM/TTL compatible. 
Which laser driver do you have?
Which version of GRBL do you have?

joncarns, 2019-10-28:
On mine the difference makes was on the actual board of the laser. I
believe there's a video on YouTube but you need to pull off the wires going
to the last from the board and under that will be 3 outputs. OOO. If yours
is in the position like this OO- then switch it to -OO
sorry for the poor description. If that didn't help I can find the links
that describe it much better

Emoslapper, 2019-10-29:
Ummm so i am sort of not using one..... could this be the root cause??

I have the Arduino Uno + CNC Shield V3 to a relay to the laser......

StuartB4, 2019-10-29:
Yes. You can make it work though by using a laser driver like this. 
[image]
Available on eBay at various prices. 

The top red circle is the Laser Voltage Adjuster, the bottom one is the Laser Current Adjuster.

The Laser Tube connections top left are for a normal 2 wire laser.

The bottom right connector is where you connect 1 wire from Z+ or Z- on the CNC Shield,
and 1 wire from a Ground pin on the shield. The Z+ and Z- are both connected together on the shield so it does'nt matter which one you use, the ground pin can be any of the row of pins in the row next to it.

Emoslapper, 2019-10-30:
Thanks for the quick reply! Ill keep you posted once the driver arrives in the mail ;)
Also wow what a great setup, nice work @StuartB4 

Thanks again

WelshHighlander, 2025-02-12:
> Maybe I could add a config switch for who does not have PWM ability that replace all S command with M3/M5 command based on threshold. I added it on dev roadmap.

I spent some years engineering industrial CNC controls but never used a consumer CNC machine.  I bought a Creality Falcon 10w and expected laser controls to work in a standard way.  Well, no.

Sxxx/M3/M4/M5 should work like this on any CNC with a non-rotating cutting tool (e.g. cutting jet, arc cutter, etc):

M3 = spindle on (laser on)
M5 = spindle off (laser off)
M4 = M3, or M4 disabled (common for non-rotating cutting tools)
Sxxx = speed (in our case laser power)

In a real CNC machine, S commands and the M3/M4/M5 commands are independent.    You should be able to set the power (Sxxx) one time and then turn the laser on and off with G3/G5 without ever having to issue another S command.  In response to G3, the laser operates at the power specified in the last S command.  This way, the gcode in lasergrbl would operate in a standard way, it would greatly simplify (and shrink) the generated gcode, it would simplify gcode editing.  Might you fix this any time soon?

Is the lasergrbl application in the Creality Falcon 10w open source?  I can fix this myself if I have sufficient access.  All the electronic hardware is in place in the Creality Falcon, so fixing it in the electronics is not the way, for a number of reasons.