arkypita/LaserGRBL #24: laser has shifted image

Gabone06, 2017-04-09:
Can anyone tell me if the laser has shifted image?
the problem is the machine that we tested on two different cnc.
[image]
[image]

arkypita, 2017-04-09:
It seems to have slipped the belt or that there was a mechanical problem (motor hang?) during engraving process.

I don't think it can depend in any way by LaserGRBL, you should observe the machine while running the job and see if something blocks / disturb the movement of the laser.

Gabone06, 2017-04-10:
I have 2 cnc (1 is made with cd-roms and 1 with nema without belt.) When i make litle things vectorized i don t have problem. The problem appears when i make pictures.

mayhem2408, 2017-04-10:
@Gabone06 I haven't seen a problem on my CoreXY rig. Do you have a gcode file you can share that you are having a problem with. I can run it on my rig and see what happens.

Gabone06, 2017-04-20:
look at this error 
[image]

arkypita, 2017-04-20:
Is it a repeatable problem? Meaning that it happens in the same place each time you try or only this one time? 

Could you try to check GCode without running motors? Grbl provide this function by [$C command](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands#c---check-gcode-mode).

[Grbl documentation](https://github.com/grbl/grbl/wiki/Interfacing-with-Grbl#grbl--help-message) say that ID 25 mean "A G-code word was repeated in the block.". i.e a line with "G1 X10 X5" will run in error 25, 

[image]

As you can see with your eyes, LaserGRBL does not produce invalid GCode lines: "G1 X1 Y-1 F2500 S0" is perfectly valid GCode line. This mean that something was happened between pc-usbport-cable-serial chip-arduino and the valid line sent from PC was interpreted as invalid from grbl on arduino, maybe because serial receiver chip confuse Y as X and generate something like G1 X1 X-1 F2500 S0

Causes could be bad shielded or too long cable, poor quality arduino chinese clone, stepper motor or power line disturbs etc etc etc

https://www.shapeoko.com/forum/viewtopic.php?t=4268&p=31945

arkypita, 2017-04-20:
https://www.shapeoko.com/forum/viewtopic.php?f=3&t=8669

mayhem2408, 2017-04-20:
@Gabone06 Looks like you might have a communication problem between your PC and your Arduino. The Gcode line where you have a reported error and the surrounding lines are perfectly valid. I'd say you are experiencing corruptions. Are you able to share the Gcode file so I can test it on my machine?

closed 2017-05-13 by arkypita