arkypita/LaserGRBL #27: Grayscale

Rost3, 2017-04-09:
Grayscale range is very small. when engraving test 8-bit colors, the difference is very small. some shades merge. is it possible to make a setting for each Grayscale color? [](https://4.bp.blogspot.com/-oVHtmF0o0JY/WBt-WXim_UI/AAAAAAAABjE/7IUjAnLQnRE1zhqliIJGXFmo-mSGEGVawCLcB/s400/%25D0%25BC%25D0%25B8%25D1%2588%25D0%25B5%25D0%25BD%25D1%258C%2B%25D1%2581%25D0%25B5%25D1%2580%25D0%25BE%25D0%25B3%25D0%25BE.bmp)

arkypita, 2017-04-10:
Hi Rost3

It is known that some materials react very differently to the laser.
I did some tests: in my results, if we talk about wood, the classic poplar plywood is less suitable for grayscale. Its behavior is totally clear between 0-30% of the laser power (with a 2W blue laser), then between 30 and 60% is brownish, and above 60% is completely burned. So is able to reflect only 3 tones.

Much better results are obtained with the plywood MDF, that is the one i use in my video: https://www.youtube.com/watch?v=conZiopJF3k&t=2s&list=PLABqghG_gwj1-62OBAymMF9-KBXPyrW0P&index=3

However "Non-linear color to power conversion" it is one of the things planned in the roadmap.

My idea is to add a curve control like the one use in image editing software to convert 0-255 image grayscale to 0-255 power values using a curve.

[image]

arkypita, 2017-04-10:
Also I suggest you try using dithering with a fairly high resolution (8.10 lines / mm). I found some very good results with this technique.

Rost3, 2017-04-10:
in my country 90% plywood  is a birch plywood. idea to add a curve control is good idea.

closed 2017-06-05 by arkypita