arkypita/LaserGRBL #28: I want to translate into another language

Rost3, 2017-04-09:
i translate many things (game/mod) to russian and ukrainian. and want to translate yours programm. but dont know how.

arkypita, 2017-04-10:
Translation into other languages is widely appreciated.
It is easy and does not require particular skills

Strings are embedded in resource files, with .resx estensions. Here there are a .zip file with latest resx files.
[Resources.zip](https://github.com/arkypita/LaserGRBL/files/909725/Resources.zip)

A good tool for editing resx is called ResEx, and could be downloaded here:
http://resex.codeplex.com/

1) install ResEx
2) unzip resources zip file
3) Launch ResEx and open a base .resx file (not the one .it or .es that are localized)
4) From menu "Tools" select "add culture"
5) Edit strings and save the file

Finally make a new .zip file and send it to me, I will compile a new version adding your languages.

Have a nice day

Rost3, 2017-04-10:
ok

closed 2017-06-05 by arkypita

Focus3, 2018-05-15:
[Added Slovenian files.zip](https://github.com/arkypita/LaserGRBL/files/2002987/Added.Slovenian.files.zip)

I did my best with the Slovenian translation. Please let me know if there is issues.

Focus3.

Mrjavaci, 2021-01-19:
> Translation into other languages is widely appreciated.
> It is easy and does not require particular skills
> 
> Strings are embedded in resource files, with .resx estensions. You can find this files inside source code, source can be downloaded by clone/download button in main github page: https://github.com/arkypita/LaserGRBL
> 
> A good tool for editing resx is called ResEx, and could be downloaded here:
> [ResExSetup1.2.zip](https://github.com/arkypita/LaserGRBL/files/909750/ResExSetup1.2.zip)
> 
> 1. install ResEx
> 2. download source
> 3. Launch ResEx and open mains .resx file (not the one .it or .es that are localized) some are in the main folder, some in subfolders, so search for *.resx
> 4. From menu "Tools" select "add culture"
> 5. Edit strings and save the file
> 6. Repeat for each .resx main file
> 
> Finally make a .zip file (of all the new added .resx files) and send it to me, I will compile a new version adding your languages.
> 
> Files involved in translation are:
> 
> * LaserGRBL\ConnectLogForm.resx
> * LaserGRBL\CustomButtonForm.resx
> * LaserGRBL\GrblConfig.resx
> * LaserGRBL\HotkeyManagerForm.resx
> * LaserGRBL\IssueDetectorForm.resx
> * LaserGRBL\JogForm.resx
> * LaserGRBL\MainForm.resx
> * LaserGRBL\PreviewForm.resx
> * LaserGRBL\ResumeJobForm.resx
> * LaserGRBL\RunFromPositionForm.resx
> * LaserGRBL\SettingsForm.resx
> * LaserGRBL\Strings.resx
> * LaserGRBL\AutoUpdate\NewVersionForm.resx
> * LaserGRBL\RasterConverter\ConvertSizeAndOptionForm.resx
> * LaserGRBL\RasterConverter\RasterToLaserForm.resx
> * LaserGRBL\RasterConverter\ResolutionHelperForm.resx
> * LaserGRBL\SvgConverter\ConvertSizeAndOptionForm.resx
> 
> Have a nice day

Are the files related to Translate still the same?

If not, what are the new files?

I am asking to add Turkish support.

arkypita, 2021-01-19:
I suggest you to use the guide you can find here: https://github.com/arkypita/LaserGRBL/blob/master/TRANSLATING.md

This requires you to use git/github to exchange translated files, it is a bit more complex at first but allows for better and more controlled source management.

arkypita, 2021-01-19:
Here is an up-to-date list of .resx file to translate

- LaserGRBL\ConnectLogForm.resx
- LaserGRBL\CustomButtonForm.resx
- LaserGRBL\ExceptionManager.resx
- LaserGRBL\GrblConfig.resx
- LaserGRBL\HotkeyManagerForm.resx
- LaserGRBL\IssueDetectorForm.resx
- LaserGRBL\JogForm.resx
- LaserGRBL\LicenseForm.resx
- LaserGRBL\MainForm.resx
- LaserGRBL\PreviewForm.resx
- LaserGRBL\ResumeJobForm.resx
- LaserGRBL\RunFromPositionForm.resx
- LaserGRBL\SaveOptionForm.resx
- LaserGRBL\SettingsForm.resx
- LaserGRBL\Strings.resx
- LaserGRBL\AutoUpdate\NewVersionForm.resx
- LaserGRBL\PSHelper\PSEditorForm.resx
- LaserGRBL\PSHelper\PSHelperForm.resx
- LaserGRBL\RasterConverter\ConvertSizeAndOptionForm.resx
- LaserGRBL\RasterConverter\RasterToLaserForm.resx
- LaserGRBL\RasterConverter\ResolutionHelperForm.resx
- LaserGRBL\SvgConverter\ConvertSizeAndOptionForm.resx