arkypita/LaserGRBL #29: multi layer configuration

mkeyno, 2017-04-19:
I've used Chines laser cutter firmware and inkscape which is has multi layer configuration and can define power and speed for each layer , do you have plan to implement such feature ?
BR

arkypita, 2017-04-19:
Hi mkeyno
For now I have no plans to implement a function of this type. The import of a raster image (the only one import implemented now) does not allow me to differentiate layers. When I will add the reading/import of vector files like .svg or .dxf I could put something like that you suggest.

best regards, Diego

mkeyno, 2017-04-19:
thanks @arkypita , you  did the hard part and I thought  with many  free C# vector2gcode lib it wont be hard to implement such feature , unfortunately I'm not C# pro to make PR in this case , but thanks for your repo , its very helpful

arkypita, 2017-04-19:
vector file import is at the top of the list of things I want to implement. It's just because of the little free time I have that I whould not been able to add it yet!

closed 2017-06-06 by arkypita

sh4wner, 2019-04-01:
Hi @arkypita, 

I´m a developer from Spain, first of all, congratulate for yout application and for bring to all people the posibility of work with it.  I past last days implementing a stroke color based multishape functionality and a material manager.

This loads and appends the SVG document N times as you mark in work repeats filtering the shapes by work selected color and adjusting the speed and power for each work.

I´ve made too  a material manager to store and load default values for each material.

I´ve not finished yet but this custom multishape functionality is working.
[image].

If you think that this can supply this feature, contact me and i push my branch wherever.

Thanks!

sh4wner, 2019-04-01:
I made too a simply and optional InkScape integration (adding Inkscape folder to settings form) to convert svg texts to svg paths through the execution of a system command like this:

inkscape filename --export-text-to-path in.svg -l out.svg

When the file is converted it replace the original filename in the rest of flux.
It works very well and it can be a solution (no desired but functional) for autoconvert svg files with text in a full paths svg. I know that this is a great dependency but for many users that have InkScape instaled can be a good point.

This is the function if any interested:
`private string preprocesTextsWithInkScape(string filename) {
            string inkScapeFolder = Settings.GetObject("InkScapeFolder", "").ToString();
            if (inkScapeFolder!="")
            {
                EnsureExistTempPathFolder();
                var outFilename = TempPath + "out.svg";
                string strCmdLine = "/C \"" + inkScapeFolder + "\\inkscape.com\" " + filename + " --export-text-to-path --export-plain-svg " + outFilename;
                try
                {
                    var process1 = System.Diagnostics.Process.Start("CMD.exe", strCmdLine);
                    process1.WaitForExit();
                    process1.Close();
                    filename = outFilename;
                }
                catch (Exception e)
                {
                }

            }

            return filename;
        }`