arkypita/LaserGRBL #30: Resolution as steps between lines and not as lines per millimeter

arkypita, 2017-04-27:
Some users have reported to have problems of rifling and grids which appear on the image

[image]

[image]

My hypothesize is that it may be due to a non-perfect coincidence between the motor step size and the required resolution. If stepper motor has a step of 0.21mm then hardware can only produce lines in this places: 0.0 - 0.21 - 0.42 - 0.63 - 0.84 - 1.05 etc. When LaserGRBL produce gcode, it does not know nothing about specific hardware, so if you ask for 5 lines/mm it produce gcode for lines at 0.0 - 0.2 - 0.4 - 0.6 - 0.8 - 1.0 etc

This mean that grbl on arduino cannot reproduce this position exactly, so it will approximate to the nearest step. This can cause some lines to be reproduced with a slightly different pitch (bigger or smaller) with a modular repetition that depends on how decimals remains will sum.

If the hardware can do something like 40lines/mm (that is very high resolution!) the approximation is soo small that cannot be seen, but if hardware has a step lower or confrontable with the resolution this could explain this behaviour.

arkypita, 2017-04-27:
In the next version of LaserGRBL i will add an option to set set resolution as steps between each line and not as lines per millimeter.

It becomes slightly more complex for the end user (who needs to know some more information about his hardware) but certainly produces more accurate results with fewer artifacts

mayhem2408, 2017-04-27:
@arkypita  I have been playing around with an inkscape plugin and some time ago I added a resolution optimisation option where you simply enter the steps per mm of the stepper motor as entered in GRBL. The resolution is then adjusted to make sure each X and Y movement are a whole number of steps. I use the following formula.

Resolution = 11 pixels/lines
StepsPermm = 160 ; 100 steps per mm as set in GRBL

            newresolution = StepsPermm / round( (StepsPermm/Resolution) ,0)

             newresolution = 160 / round( (160/11) ,0 )
             newresolution = 160 / round ( 14.545454545455 , 0 )
             newresolution = 160 / 15
             newresolution = 10.666666666667

So with a new resolution of 10.666666667 lines per milimeter, you will get a perfect 15 steps between each pixel. 

Hope you can use this formula.

dungjk, 2017-04-28:
@arkypita I think that we can balancing between easy to use and precision. We can allow user to input float number instead of int number (with default int number). It's little bit more complex for user but it can resolve user issues with some guide.
I'm planning to write a wiki page to help user guess minimum resolution of there machine.

arkypita, 2017-04-28:
I'm thinking to get the resolution from arduino by reading $100, $101 parameters and store them in LaserGRBL (so that they can be used without having to re-read each time from device).

Then provide the choice between a range of resolutions that can be managed by hardware, as multiples of the minimum pitch.

arkypita, 2017-05-01:
Implemented and solved in [v2.6.0](https://github.com/arkypita/LaserGRBL/releases/tag/v2.6.0)

closed 2017-05-01 by arkypita

arkypita, 2017-05-01:
@dungjk if you want to write i wiki for image resolution, with instructions and sample images, i will put your job here: http://lasergrbl.com/usage/raster-image-import/setting-reliable-resolution/

mayhem2408, 2017-05-02:
@arkypita  The resolution helper doesn't seem to have the desired effect. If I have steps per mm at 80 and the desired resolution at 11 lines per mm, the computed resolution is still 11. However 11 is 7.27 steps. If you drop it back to 7 steps per line, the resolution should be 11.4286.

arkypita, 2017-05-02:
New version: https://github.com/arkypita/LaserGRBL/releases/tag/v2.6.1

mayhem2408, 2017-05-02:
Awesome. 2.6.1 works OK, but can you also increase the steps to mm limit from 100 to maybe 200. I have 5 machines and 2 of them are above your maximum of 100.

1. Mini CD Rom bases 2 Axis @0.5W with **53.3333** Steps per mm.
2. DIY 2 Axis @ 3.5W with **80** Steps per mm.
3. DIY 2 Axis CoreXY @ 5W with **100** Steps per mm.
4. Stepcraft 2 with DIY Laser @ 5W running **133.3333** Steps per mm.
5. K40 CO2 Laser with **160** Steps per mm.

arkypita, 2017-05-02:
Of course I can, I did not think there was HW with such high resolutions :-)

mayhem2408, 2017-05-02:
@arkypita Awesome. Thanks. As far as high resolution goes, My Stepcraft machine has a resolution of 133.3333 steps per mm and the stepper drivers are only 1/2 step microstepping. If I set them to 1/16, it would be 1066.66666 steps per mm. But that's just overkill and GRBL can only handle 30,000 steps per second which would limit my feed to about F1800. Seems pointless having that high a resolution and sacrificing so much speed.

mayhem2408, 2017-05-03:
@arkypita V2.6.2 works great. I'll post some pictures shortly. My inkscape plugin is becoming more and more redundant as this code evolves. Great work.

emilvv, 2018-05-25:
Hello,

May I ask how to setup quality to 30 lines / mm - I have last LaserGRBL and my setup is limiting me only to 20 lines / mm ....

I have UV laser diode and real track is about 0.03 / 0.04 mm on my PCB UV printer ..

Thank  you in advance,

Emil

arkypita, 2018-05-28:
@emilvv 
It is easy for me to remove the limit of 20 line but LaserGRBL was not designed for engraving PCB and for very small output size, so the limit of 20 is not the only one... for example another limit is the needs of integer value of engraving size (and others hardcoded limits) 

I know that there are a lot of people using LaserGRBL for PCB's but to made LaserGRBL good for this task is it necessary a big code rethink that i cannot make now

emilvv, 2018-05-28:
Hello,

Thank you for your answer !

Is it possible to use another g-code generator with step about 0.04 / mm and then to upload it to lasergrbl and execute it ?? It is in case that hardware is supporting this resolution ...

Regards,

Emil

arkypita, 2018-05-28:
> Is it possible to use another g-code generator with step about 0.04 / mm and then to upload it to lasergrbl and execute it ??

Yes, lasergrbl is basically a G-Code streamer and its main use is to load and send a gcode text file to your controller (usually an arduino with grbl firmware).

The raster import/vectorization is a secondary feature for who needs a simple way to engrave a picture and does not claim to be able of compete with "precision" task like engraving technical drawings, pdb & co that require a very fine control of generated gcode.

In my opinion starting from a vector information such essentially is the design of electric tracks, transform it into a raster jpg image, and then re-transform it into lines and paths with a vectorization operation can certainly not produce reliable results in terms of precision of lines and holes.

The better solution is to have some software than can generate gcode from the original file format of your electronic cad but I don't know if exists

emilvv, 2018-05-28:
Hello,

Because UV PCB printing is 'filling' PCBs tracks areas .... it is raster engraving ...

Row by row ....and it is much faster than vector ... engraving ..

Example video ..   https://www.youtube.com/watch?v=l4BX2r0ZkjI

Thank you for your answer ..

Emil