arkypita/LaserGRBL #31: bluetooth  serial bridge can't establish connection

mkeyno, 2017-04-30:
hi @arkypita , I use bluetooth module to made serial bridge between hardware and station , I have no problem with all gcode sender except your software ,it can't establish connection , have any idea what might be problem?

arkypita, 2017-04-30:
I think you could have the same problem as #32 

To help me solving this Issue you must provide me the exact grbl version, as reported with `$I` command and the value of `$10` status report mask settings. Even better if you can tell me how is the string of realtime status that grbl returns on `?` command

Could you also try without bluoetooth (so with direct USB cable) to see if there is any difference

mkeyno, 2017-04-30:
I tested both .9j and 1.1f , none of combination did not connect through bluetooth serial bridge  while its work fine with usb serial (cable) connection , FYI , $10=0 	(Status report options, mask)
As it  stated before your software unlike the other gcode sender can't made wireless serial bridge

arkypita, 2017-05-01:
Ok! So you're telling me that LaserGRBL works well if it is connected via cable but it does not work via bluetooth, while other software works properly both via cable and bluetooth. It's correct?

So the problem then resides in how .net (the system with which LaserGRBL is programmed) handles COM/Bluetooth ports.

Can you give me some additional information about this bluetooth connection? Which bluetooth module do you use on the arduino side? And on the PC side, do you use integrated bluetooth or do you have an external module? Do you use a specific driver/software to see bluetooth as a COM port or is it completely transparent/automatic in windows? LaserGRBL shows the right COM port in port list?

Since this is a problem I can hardly reproduce, because I do not have a bluetooth module to try, what I can do is write a simple diagnostic program and send it to you.

I am googling about issue with .net comport and bluetooth but without knowing the specific model of your bluetooth adapter the search can not be very effective. 

www.google.it/search?client=opera&q=.net+comport+bluetooth
www.google.it/search?client=opera&q=.net+comport+bluetooth+issues
www.google.it/search?client=opera&q=.net+comport+bluetooth+bug

mkeyno, 2017-05-01:
I use HC-6 to control my Arduino modules, its paired to my laptop BT by Standard Serial over Bluetooth protocol and hence I could easily open it by program like Termite or realterm 

http://2.bp.blogspot.com/-ANqBbReXkt4/T99Bw9aJWDI/AAAAAAAAAI0/eBYdC4Yr5Eo/s1600/HC-06.png

arkypita, 2017-05-01:
I have published a new version of LaserGRBL that records any connection problem and internal program errors in a session log file. 

https://github.com/arkypita/LaserGRBL/releases/tag/v2.6.0

Please update to the new version, try connect LaserGRBL (perform 2-3 attempts) and open session log file. 

[image]

Then copy and paste your session log here.

mkeyno, 2017-05-02:
thanks @arkypita for your concern , I install new software , and here its feedback
```
5/2/2017 11:31:53 AM	SetStatus   	Machine status [Connecting]
5/2/2017 11:31:53 AM	OpenCom     	Open COM11ʕ @ 115200 baud
5/2/2017 11:31:53 AM	OpenCom     	System.IO.IOException: The port 'COM11ʕ' does not exist.
5/2/2017 11:31:53 AM	OpenCom     	   at System.IO.Ports.InternalResources.WinIOError(Int32 errorCode, String str)
5/2/2017 11:31:53 AM	OpenCom     	   at System.IO.Ports.SerialStream..ctor(String portName, Int32 baudRate, Parity parity, Int32 dataBits, StopBits stopBits, Int32 readTimeout, Int32 writeTimeout, Handshake handshake, Boolean dtrEnable, Boolean rtsEnable, Boolean discardNull, Byte parityReplace)
5/2/2017 11:31:53 AM	OpenCom     	   at System.IO.Ports.SerialPort.Open()
5/2/2017 11:31:53 AM	OpenCom     	   at LaserGRBL.GrblCore.OpenCom()
5/2/2017 11:32:09 AM	SetStatus   	Machine status [Disconnected]
```
as you see the Bluetooth COM number came with suffix letter '?' , I don't know why? 
[image]

arkypita, 2017-05-02:
Googling about bluetooth and name of the function I use to read available comport names (getportnames) show [a lot of issue signaled by the community](www.google.it/search?client=opera&q=system.io.ports.serialport.getportnames+error+with+bluetooth). It seems to be a bug of .NET framework under WindowsXP

One of those result suggest some tricks to solve the problems. Some tricks is on program side (cleanup port names) some on user side (a regedit trick).

http://stackoverflow.com/questions/32040209/serialport-getportnames-returns-incorrect-port-names

If you want to test the regedit way follow this guide:

Go in to the Registry with regedit.exe.
1) Navigate to "HKEY_LOCAL_MACHINE\HARDWARE\DEVICEMAP\SERIALCOMM"
2) Make note of the comm port name.
3) Append a "0" or any character to the comm port name.
4) Change the comm port name back to what it was in step 3.

mkeyno, 2017-05-02:
@arkypita I have last publish of windows 7 64bit, its nothing related to windows xp ,following is simple USB communication which is work for both  local USB serial and Bluetooth serial , as it mentioned COM port name appeared with extra character which is let to unknown for systems.IO. routine handler, may I suggest to made adhoc solution to remove the last char before open the port  and test it again 

```cpp
using System;
using System.Collections.Generic;
using System.ComponentModel;
using System.Data;
using System.Drawing;
using System.Text;
using System.Windows.Forms;
using System.IO;
using System.IO.Ports;

namespace Usb_Reader
{
    public partial class Form1 : Form
    {
        char oldString = '\0';
        
        StreamReader sr;
        public Form1()
        {
            InitializeComponent();
            comboBox1.Items.AddRange(SerialPort.GetPortNames());
        }

        private void btnconnect_Click(object sender, EventArgs e)
        {
            if (btnconnect.Text == "Connect")
            {
                if (comboBox1.SelectedIndex != -1)
                {
                    serialPort1.PortName = comboBox1.SelectedItem.ToString();
                    try
                    {
                        serialPort1.Open();
                        btnconnect.Text = "DisConnect";
                        panel1.Enabled = true;
                        timer1.Enabled = true;
                        try
                        {
                            timer1.Interval = int.Parse(interval.Text);
                        }
                        catch(Exception)
                        {
                            timer1.Interval = 100;
                        }
                    }
                    catch (Exception ex)
                    {
                        MessageBox.Show(ex.Message);
                    }

                }
            }
            else
            {
                btnconnect.Text = "Connect";
                serialPort1.Close();
                panel1.Enabled = false;
                timer1.Enabled = false;
            }
        }

        private void txtwrite_KeyPress(object sender, KeyPressEventArgs e)
        {

            if (e.KeyChar.ToString() == "\r")
            {
                serialPort1.WriteLine(txtwrite.Text);
            }
        }

        private void serialPort1_DataReceived(object sender, SerialDataReceivedEventArgs e)
        {
            string data = serialPort1.ReadLine();
            if (data == "\r")
            {
                return;
            }
            StreamWriter sw = new StreamWriter("usb_in.txt");
            
            sw.Write(data);
            if (txtread.InvokeRequired)
            {
                txtread.Invoke(new MethodInvoker(delegate { txtread.Text = data; }));
            }
            sw.Close();
            
        }
        
        private void timer1_Tick(object sender, EventArgs e)
        {
            try
            {
                timer1.Interval = int.Parse(interval.Text);
            }
            catch (Exception)
            {
                timer1.Interval = 100;
            }
            File.Copy("php_in.txt", "php_in_Copy.txt",true);
            sr= new StreamReader("php_in_Copy.txt");
            string line = sr.ReadLine();
            if (line[0] != oldString)
            {
                txtwrite.Text = line;
                //if (txtwrite.InvokeRequired)
                //{
                //    txtwrite.Invoke(new MethodInvoker(delegate { txtwrite.Text = line; }));
                //}
                serialPort1.WriteLine(line);
                oldString = line[0];
            }
            sr.Close();
        }

        private void interval_KeyPress(object sender, KeyPressEventArgs e)
        {
            //int result = 0;
            //e.Handled = int.TryParse(interval.Text, out result);
        }
    }
}

```

arkypita, 2017-05-02:
GetPortNames() read port names from registry keys located in HKEY_LOCAL_MACHINE\HARDWARE\DEVICEMAP\SERIALCOMM

Sometimes Windows has wrong names in this list. If the names in this list are wrong GetPortNames returns them obviously wrong.

The best solution is to clean up the names within the program, as the ability to edit and fix registry requires advanced skills, and it is not definitive.

Wrong names has a single random final character in excess of the due. Since random character means that it could also be a numerical char, the simple solution of removing all non-numeric final characters can not cover all the cases.

Remove non numerical should be a first pass.
The second pass is to use a fallback procedure that intercept `System.IO.IOException: The port 'COM11ʕ' does not exist.` and try again removing one char, as you suggest.

I will publish a fix for you as soon as possible
Thanks for your feedback 👍

mkeyno, 2017-05-02:
thanks for your concern, but as you can see the port name is correct (as it expected) so it must another reason which is made GetPortNames returns wrong list
[image]

arkypita, 2017-05-02:
You see it as correct, but maybe it is not.
I think that the binary data behind the string it is not well terminated, so when copied between registry and GetPortNames() the function does not know the string lenght.

Just try to edit the string "COM11" in something like "FOO" and save, then edit back "FOO" to "COM11" and save it again. Now the string should be ok.

Then run or restart LaserGRBL, and see if you have some difference... I am just curious to see if it fix!

mkeyno, 2017-05-02:
your first solution its not work but when I edit  back to original name , it went ok as following log , I suggest change property of list box to editable so I could edit the port name in such incident 

```
5/2/2017 6:54:32 PM	SetStatus   	Machine status [Disconnected]
5/2/2017 6:54:34 PM	SetStatus   	Machine status [Connecting]
5/2/2017 6:54:34 PM	OpenCom     	Open FOO @ 115200 baud
5/2/2017 6:54:34 PM	OpenCom     	System.ArgumentException: The given port name does not start with COM/com or does not resolve to a valid serial port.
5/2/2017 6:54:34 PM	OpenCom     	Parameter name: portName
5/2/2017 6:54:34 PM	OpenCom     	   at System.IO.Ports.SerialStream..ctor(String portName, Int32 baudRate, Parity parity, Int32 dataBits, StopBits stopBits, Int32 readTimeout, Int32 writeTimeout, Handshake handshake, Boolean dtrEnable, Boolean rtsEnable, Boolean discardNull, Byte parityReplace)
5/2/2017 6:54:34 PM	OpenCom     	   at System.IO.Ports.SerialPort.Open()
5/2/2017 6:54:34 PM	OpenCom     	   at LaserGRBL.GrblCore.OpenCom()
5/2/2017 6:55:22 PM	SetStatus   	Machine status [Disconnected]
5/2/2017 6:55:26 PM	Program     	------------ PROGRAM STOP ------------

5/2/2017 6:55:33 PM	Program     	------------ PROGRAM START ------------
5/2/2017 6:55:35 PM	SetStatus   	Machine status [Disconnected]
5/2/2017 6:55:39 PM	SetStatus   	Machine status [Connecting]
5/2/2017 6:55:39 PM	OpenCom     	Open COM11 @ 115200 baud
5/2/2017 6:55:41 PM	SendImmediate	Send Immediate Command [24]
5/2/2017 6:55:41 PM	SetStatus   	Machine status [Idle]

```

arkypita, 2017-05-02:
Of course "FOO" must not work! "FOO" it is a string like any other (FOO, PIPPO, YOURNAME. John, 1234) and it is used only to change COM11 in something else and save... not to test it :-)

The original COM11 stored in registry was stored wrong, also if you saw COM11. 

The sequence edit as "foo" -> save -> edit as "COM11" -> save it again, is to be sure to save COM11 as new string and overwrite the old dummy value.

mkeyno, 2017-05-02:
thanks @arkypita  your temporary solution work now , however after log to software and try to engrave some test , its freeze as  code in following image and only need hardware reset , its may not related to software and I don't want to hijack the ticket , so if you disagree let me know to clear this ticket 
[image]
 firmware setting 
```
> $$ 
< $0=44 	(Step pulse time, microseconds)
< $1=49 	(Step idle delay, milliseconds)
< $2=27 	(Step pulse invert, mask)
< $3=5 	(Step direction invert, mask)
< $4=1 	(Invert step enable pin, boolean)
< $5=0 	(Invert limit pins, boolean)
< $6=0 	(Invert probe pin, boolean)
< $10=0 	(Status report options, mask)
< $11=0.000 	(Junction deviation, millimeters)
< $12=-2147483.648 	(Arc tolerance, millimeters)
< $13=1 	(Report in inches, boolean)
< $20=0 	(Soft limits enable, boolean)
< $21=1 	(Hard limits enable, boolean)
< $22=1 	(Homing cycle enable, boolean)
< $23=49 	(Homing direction invert, mask)
< $24=2147483.648 	(Homing locate feed rate, mm/min)
< $25=2147483.648 	(Homing search seek rate, mm/min)
< $26=250 	(Homing switch debounce delay, milliseconds)
< $27=0.000 	(Homing switch pull-off distance, millimeters)
< $30=0 	(Maximum spindle speed, RPM)
< $31=0 	(Minimum spindle speed, RPM)
< $32=1 	(Laser -mode enable, boolean)
< $100=-2147483.648 	(X -axis steps per millimeter)
< $101=0.000 	(Y -axis steps per millimeter)
< $102=0.000 	(Z -axis steps per millimeter)
< $110=0.000 	(X -axis maximum rate, mm/min)
< $111=0.000 	(Y -axis maximum rate, mm/min)
< $112=-0.000 	(Z -axis maximum rate, mm/min)
< $120=0.000 	(X -axis acceleration, mm/sec^2)
< $121=0.000 	(Y -axis acceleration, mm/sec^2)
< $122=-0.000 	(Z -axis acceleration, mm/sec^2)
< $130=2147483.648 	(X -axis maximum travel, millimeters)
< $131=-0.000 	(Y -axis maximum travel, millimeters)
< $132=-0.000 	(Z -axis maximum travel, millimeters)
< ok
```

arkypita, 2017-05-02:
Version v2.6.2 fix this bug/issue

https://github.com/arkypita/LaserGRBL/releases/tag/v2.6.2

closed 2017-05-02 by arkypita

arkypita, 2017-05-15:
I have a question for you @mkeyno 

If I want to add a bluetooth module in my engraver, i could simple buy a arduino module like this one (see link) and connect power, tx and rx and that's all, or I must write some code inside arduino grbl sketch to initialize the bluetooth board?

http://www.ebay.it/itm/Modulo-HC-06-scheda-bluetooth-4-pin-wireless-RF-transceiver-module-arduino-/222008015892?hash=item33b0b57c14:g:VK8AAOSwpzdWqTMZ

mkeyno, 2017-05-16:
@arkypita  just connect the pins , no extra programing needed , that's why it called bridge , however I intend to use  ESP8266 instead of BT , because its had more feature than simple serial transparent , the same in [laseweb4 ](https://github.com/LaserWeb/LaserWeb4), it used [websocket server on ESP side](https://github.com/LaserWeb/LaserWeb3-ESP8266) to create serial bridge , I was wondering if you have any future plan for this , I  think its need websocket client on your software  such as following lib 

https://websocket4net.codeplex.com/
https://github.com/sta/websocket-sharp
https://msdn.microsoft.com/en-us/library/system.net.websockets.clientwebsocket(v=vs.110).aspx?cs-save-lang=1&cs-lang=csharp#code-snippet-1
http://stackoverflow.com/questions/30523478/connecting-to-websocket-using-c-sharp-i-can-connect-using-javascript-but-c-sha
https://codereview.stackexchange.com/questions/41591/websockets-client-code-and-making-it-production-ready

BTW, of the topic , there is little exception when internet connection is not available 
[image]

debug 
```
System.Net.WebException was unhandled
  Message=The remote name could not be resolved: 'lasergrbl.com'
  Source=System
  StackTrace:
       at System.Net.WebClient.DownloadDataInternal(Uri address, WebRequest& request)
       at System.Net.WebClient.DownloadString(Uri address)
       at System.Net.WebClient.DownloadString(String address)
       at LaserGRBL.GitHub.CheckSite(String site)
       at LaserGRBL.GitHub.AsyncCheckVersion(Object foo)
       at System.Threading._ThreadPoolWaitCallback.WaitCallback_Context(Object state)
       at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
       at System.Threading._ThreadPoolWaitCallback.PerformWaitCallbackInternal(_ThreadPoolWaitCallback tpWaitCallBack)
       at System.Threading._ThreadPoolWaitCallback.PerformWaitCallback(Object state)
  InnerException: 

```

arkypita, 2017-05-17:
thanks for infos and hint
i have just ordered my bluetooth HC6 module :-)

also bug fixed in latest version