arkypita/LaserGRBL #32: Won't connect to UNO

chbarnes3, 2017-04-30:
Just loaded laserGRBL and it will not connect to Arduino UNO with the GRBLShield. Arduino's program and all others I've tried connect without issue.

arkypita, 2017-04-30:
LaserGRBL connection sequence is:
- Start [Disconnected]
- Click on connect button [Connecting]
- Receive & parse a status messages from arduino to know the right status [Idle, Run, Hold, Door, Home, Alarm, Check, Jog]

If LaserGRBL has an error in decoding status message, the effect could be that stuck in "connecting" state.

LaserGRBL assume that status message could be in one of this two format:
"<Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>" **for grbl v0.9**
"<Idle|MPos:0.000,0.000,0.000|FS:0.0,0>" **for grbl v1.1**

To help me solving this Issue you must provide me the exact grbl version, as reported with `$I` command and the value of `$10` status report mask settings.

Even better if you can tell me how is the string of realtime status that grbl returns on `?` command

arkypita, 2017-05-01:
I have published a new version of LaserGRBL that records any connection problem and internal program errors in a session log file. 

https://github.com/arkypita/LaserGRBL/releases/tag/v2.6.0

Please update to the new version, try connect LaserGRBL (perform 2-3 attempts) and open session log file. 

[image]

Then copy and paste your session log here.

arkypita, 2017-05-25:
user not responding... i close the issue

closed 2017-05-25 by arkypita