arkypita/LaserGRBL #33: can open image with china language  win7 system

jumbo5566, 2017-05-04:
2017/5/4 12:52:18	Program     	------------ PROGRAM START ------------
2017/5/4 12:52:20	SetStatus   	Machine status [Disconnected]
2017/5/4 12:52:21	SetStatus   	Machine status [Connecting]
2017/5/4 12:52:21	OpenCom     	Open COM9 @ 115200 baud
2017/5/4 12:52:21	SendImmediate	Send Immediate Command [24]
2017/5/4 12:52:22	SetStatus   	Machine status [Idle]
2017/5/4 12:54:20	OpenFile    	Open C:\Users\allen\Downloads\激光雕刻用29.nc
2017/5/4 12:55:02	OpenFile    	Open C:\Users\allen\Downloads\1514787301.jpg
2017/5/4 12:55:02	RasterImport	System.NullReferenceException: 未将对象引用设置到对象的实例。
2017/5/4 12:55:02	RasterImport	   在 LaserGRBL.RasterConverter.RasterToLaserForm.PbConverted_Resize(Object sender, EventArgs e)
2017/5/4 12:55:02	RasterImport	   在 System.Windows.Forms.Control.OnResize(EventArgs e)
2017/5/4 12:55:02	RasterImport	   在 System.Windows.Forms.PictureBox.OnResize(EventArgs e)
2017/5/4 12:55:02	RasterImport	   在 System.Windows.Forms.Control.OnSizeChanged(EventArgs e)
2017/5/4 12:55:02	RasterImport	   在 System.Windows.Forms.Control.UpdateBounds(Int32 x, Int32 y, Int32 width, Int32 height, Int32 clientWidth, Int32 clientHeight)
2017/5/4 12:55:02	RasterImport	   在 System.Windows.Forms.Control.UpdateBounds(Int32 x, Int32 y, Int32 width, Int32 height)
2017/5/4 12:55:02	RasterImport	   在 System.Windows.Forms.Control.SetBoundsCore(Int32 x, Int32 y, Int32 width, Int32 height, BoundsSpecified specified)
2017/5/4 12:55:02	RasterImport	   在 System.Windows.Forms.Control.ScaleControl(SizeF factor, BoundsSpecified specified)
2017/5/4 12:55:02	RasterImport	   在 System.Windows.Forms.Control.ScaleControl(SizeF includedFactor, SizeF excludedFactor, Control requestingControl)
2017/5/4 12:55:02	RasterImport	   在 System.Windows.Forms.Control.Scale(SizeF includedFactor, SizeF excludedFactor, Control requestingControl)
2017/5/4 12:55:02	RasterImport	   在 System.Windows.Forms.Control.ScaleChildControls(SizeF includedFactor, SizeF excludedFactor, Control requestingControl)
2017/5/4 12:55:02	RasterImport	   在 System.Windows.Forms.Control.Scale(SizeF includedFactor, SizeF excludedFactor, Control requestingControl)
2017/5/4 12:55:02	RasterImport	   在 System.Windows.Forms.Control.ScaleChildControls(SizeF includedFactor, SizeF excludedFactor, Control requestingControl)
2017/5/4 12:55:02	RasterImport	   在 System.Windows.Forms.Control.Scale(SizeF includedFactor, SizeF excludedFactor, Control requestingControl)
2017/5/4 12:55:02	RasterImport	   在 System.Windows.Forms.Control.ScaleChildControls(SizeF includedFactor, SizeF excludedFactor, Control requestingControl)
2017/5/4 12:55:02	RasterImport	   在 System.Windows.Forms.Control.Scale(SizeF includedFactor, SizeF excludedFactor, Control requestingControl)
2017/5/4 12:55:02	RasterImport	   在 System.Windows.Forms.ContainerControl.OnChildLayoutResuming(Control child, Boolean performLayout)
2017/5/4 12:55:02	RasterImport	   在 System.Windows.Forms.Control.OnLayoutResuming(Boolean performLayout)
2017/5/4 12:55:02	RasterImport	   在 System.Windows.Forms.Control.ResumeLayout(Boolean performLayout)
2017/5/4 12:55:02	RasterImport	   在 LaserGRBL.RasterConverter.RasterToLaserForm.InitializeComponent()
2017/5/4 12:55:02	RasterImport	   在 LaserGRBL.RasterConverter.RasterToLaserForm..ctor(GrblCore core, String filename)
2017/5/4 12:55:02	RasterImport	   在 LaserGRBL.RasterConverter.RasterToLaserForm.CreateAndShowDialog(GrblCore core, String filename, Form parent)
2017/5/4 12:55:02	RasterImport	   在 LaserGRBL.GrblCore.OpenFile(Form parent)

arkypita, 2017-05-04:
Whould you like to zip those file/files and upload them here, so i can try opening them and reproduce the issue?

arkypita, 2017-05-08:
same as #34

arkypita, 2017-05-08:
fixed

closed 2017-05-08 by arkypita