arkypita/LaserGRBL #34: cant open raster image import

austriandummy, 2017-05-07:
sorry, im a absolute newbie and my englisch is very bad.
mayby anyone could help me........i dont no how to integrate or install all those cool featers in lasergrbl...???

arkypita, 2017-05-08:
- File -> Open File
- Select an image file (.jpg .png .bmp...)

austriandummy, 2017-05-08:
this ones i know....but lasergrbl dosn´t open a window or import picture to work with it.

arkypita, 2017-05-08:
Did you see the open file dialog, but when you select the graphic file the raster image dialog did not show?

1) If you don't have latest version (2.6.2) please update to the new version.
2) Open the file dialog and select image file.
3) Open session log and copy and paste your session log here, let me see if there is some error in file reading.

[image]

austriandummy, 2017-05-08:
07.05.2017 17:16:00	Program     	------------ PROGRAM START ------------
07.05.2017 17:16:02	SetStatus   	Machine status [Disconnected]
07.05.2017 17:16:12	OpenFile    	Open C:\Users\Public\Pictures\Sample Pictures\Koala.jpg
07.05.2017 17:16:12	RasterImport	System.NullReferenceException: Der Objektverweis wurde nicht auf eine Objektinstanz festgelegt.
07.05.2017 17:16:12	RasterImport	   bei LaserGRBL.RasterConverter.RasterToLaserForm.PbConverted_Resize(Object sender, EventArgs e)
07.05.2017 17:16:12	RasterImport	   bei System.Windows.Forms.Control.OnResize(EventArgs e)
07.05.2017 17:16:12	RasterImport	   bei System.Windows.Forms.PictureBox.OnResize(EventArgs e)
07.05.2017 17:16:12	RasterImport	   bei System.Windows.Forms.Control.OnSizeChanged(EventArgs e)
07.05.2017 17:16:12	RasterImport	   bei System.Windows.Forms.Control.UpdateBounds(Int32 x, Int32 y, Int32 width, Int32 height, Int32 clientWidth, Int32 clientHeight)
07.05.2017 17:16:12	RasterImport	   bei System.Windows.Forms.Control.UpdateBounds(Int32 x, Int32 y, Int32 width, Int32 height)
07.05.2017 17:16:12	RasterImport	   bei System.Windows.Forms.Control.SetBoundsCore(Int32 x, Int32 y, Int32 width, Int32 height, BoundsSpecified specified)
07.05.2017 17:16:12	RasterImport	   bei System.Windows.Forms.Control.ScaleControl(SizeF factor, BoundsSpecified specified)
07.05.2017 17:16:12	RasterImport	   bei System.Windows.Forms.Control.ScaleControl(SizeF includedFactor, SizeF excludedFactor, Control requestingControl)
07.05.2017 17:16:12	RasterImport	   bei System.Windows.Forms.Control.Scale(SizeF includedFactor, SizeF excludedFactor, Control requestingControl)
07.05.2017 17:16:12	RasterImport	   bei System.Windows.Forms.Control.ScaleChildControls(SizeF includedFactor, SizeF excludedFactor, Control requestingControl)
07.05.2017 17:16:12	RasterImport	   bei System.Windows.Forms.Control.Scale(SizeF includedFactor, SizeF excludedFactor, Control requestingControl)
07.05.2017 17:16:12	RasterImport	   bei System.Windows.Forms.Control.ScaleChildControls(SizeF includedFactor, SizeF excludedFactor, Control requestingControl)
07.05.2017 17:16:12	RasterImport	   bei System.Windows.Forms.Control.Scale(SizeF includedFactor, SizeF excludedFactor, Control requestingControl)
07.05.2017 17:16:12	RasterImport	   bei System.Windows.Forms.Control.ScaleChildControls(SizeF includedFactor, SizeF excludedFactor, Control requestingControl)
07.05.2017 17:16:12	RasterImport	   bei System.Windows.Forms.Control.Scale(SizeF includedFactor, SizeF excludedFactor, Control requestingControl)
07.05.2017 17:16:12	RasterImport	   bei System.Windows.Forms.ContainerControl.OnChildLayoutResuming(Control child, Boolean performLayout)
07.05.2017 17:16:12	RasterImport	   bei System.Windows.Forms.Control.OnLayoutResuming(Boolean performLayout)
07.05.2017 17:16:12	RasterImport	   bei System.Windows.Forms.Control.ResumeLayout(Boolean performLayout)
07.05.2017 17:16:12	RasterImport	   bei LaserGRBL.RasterConverter.RasterToLaserForm.InitializeComponent()
07.05.2017 17:16:12	RasterImport	   bei LaserGRBL.RasterConverter.RasterToLaserForm..ctor(GrblCore core, String filename)
07.05.2017 17:16:12	RasterImport	   bei LaserGRBL.RasterConverter.RasterToLaserForm.CreateAndShowDialog(GrblCore core, String filename, Form parent)
07.05.2017 17:16:12	RasterImport	   bei LaserGRBL.GrblCore.OpenFile(Form parent)
07.05.2017 17:16:19	OpenFile    	Open C:\Users\Public\Pictures\Sample Pictures\Koala.jpg
07.05.2017 17:16:19	RasterImport	System.NullReferenceException: Der Objektverweis wurde nicht auf eine Objektinstanz festgelegt.
07.05.2017 17:16:19	RasterImport	   bei LaserGRBL.RasterConverter.RasterToLaserForm.PbConverted_Resize(Object sender, EventArgs e)
07.05.2017 17:16:19	RasterImport	   bei System.Windows.Forms.Control.OnResize(EventArgs e)
07.05.2017 17:16:19	RasterImport	   bei System.Windows.Forms.PictureBox.OnResize(EventArgs e)
07.05.2017 17:16:19	RasterImport	   bei System.Windows.Forms.Control.OnSizeChanged(EventArgs e)
07.05.2017 17:16:19	RasterImport	   bei System.Windows.Forms.Control.UpdateBounds(Int32 x, Int32 y, Int32 width, Int32 height, Int32 clientWidth, Int32 clientHeight)
07.05.2017 17:16:19	RasterImport	   bei System.Windows.Forms.Control.UpdateBounds(Int32 x, Int32 y, Int32 width, Int32 height)
07.05.2017 17:16:19	RasterImport	   bei System.Windows.Forms.Control.SetBoundsCore(Int32 x, Int32 y, Int32 width, Int32 height, BoundsSpecified specified)
07.05.2017 17:16:19	RasterImport	   bei System.Windows.Forms.Control.ScaleControl(SizeF factor, BoundsSpecified specified)
07.05.2017 17:16:19	RasterImport	   bei System.Windows.Forms.Control.ScaleControl(SizeF includedFactor, SizeF excludedFactor, Control requestingControl)
07.05.2017 17:16:19	RasterImport	   bei System.Windows.Forms.Control.Scale(SizeF includedFactor, SizeF excludedFactor, Control requestingControl)
07.05.2017 17:16:19	RasterImport	   bei System.Windows.Forms.Control.ScaleChildControls(SizeF includedFactor, SizeF excludedFactor, Control requestingControl)
07.05.2017 17:16:19	RasterImport	   bei System.Windows.Forms.Control.Scale(SizeF includedFactor, SizeF excludedFactor, Control requestingControl)
07.05.2017 17:16:19	RasterImport	   bei System.Windows.Forms.Control.ScaleChildControls(SizeF includedFactor, SizeF excludedFactor, Control requestingControl)
07.05.2017 17:16:19	RasterImport	   bei System.Windows.Forms.Control.Scale(SizeF includedFactor, SizeF excludedFactor, Control requestingControl)
[... cut ...]
07.05.2017 17:16:44	RasterImport	   bei LaserGRBL.RasterConverter.RasterToLaserForm..ctor(GrblCore core, String filename)
07.05.2017 17:16:44	RasterImport	   bei LaserGRBL.RasterConverter.RasterToLaserForm.CreateAndShowDialog(GrblCore core, String filename, Form parent)
07.05.2017 17:16:44	RasterImport	   bei LaserGRBL.GrblCore.OpenFile(Form parent)
07.05.2017 17:16:55	Program     	------------ PROGRAM STOP ------------

08.05.2017 18:08:44	Program     	------------ PROGRAM START ------------
08.05.2017 18:08:46	SetStatus   	Machine status [Disconnected]

arkypita, 2017-05-08:
This was a bug, I create a new LaserGRBL version with a fix.
Would you like to try and tell me if now works?

austriandummy, 2017-05-08:
shure i would :) where i can found it?

arkypita, 2017-05-08:
https://github.com/arkypita/LaserGRBL/releases

austriandummy, 2017-05-08:
i´ve got it
thanks ark, it works ;)
just waiting for my laser driver, but if my cnc runs i´ll show you some succsess with my projekt;)

arkypita, 2017-05-08:
Thank you too, you contribute solving a bug

closed 2017-05-08 by arkypita