arkypita/LaserGRBL #35: S-min Disabled

Srinath25, 2017-05-10:
I've been using this software for some days now and I notice that I can't change the S-min settings for PMW control of the laser.
I'am sure the option was available previously.

Srinath25, 2017-05-10:
[image]

arkypita, 2017-05-11:
You are right, I have removed S-MIN option.
This is because of the speed optimization feature (move fast and skip all parts with zero laser power) that cannot be effective if S-MIN > 0

Do you use S-MIN values different from zero? Why?
Maybe you can obtain same results "darkening" your image using brightness/contrast control.

Srinath25, 2017-05-11:
Yes I use it, the material I'm working with is white plywood and to get good greyscale picture on, it needs S-MIN about 40, otherwise it won't even make a scratch (50 it starts to engrave).
Also the S-MAX value can't be higher than 120 which is burning the surface.

So without the S-min it would just come out as a black and white picture with no shades.

arkypita, 2017-05-11:
Yes, I understand what you mean.
I am thinking to a solution to put together speed optimization and variable S-MIN
Meanwhile you can continue using an older version like 2.5.0 that has free control on S-MIN

closed 2017-05-13 by arkypita

arkypita, 2017-05-13:
Fixed in v2.6.5

commit cf75b5c3 (2017-05-13): Add of a white threshold

add a white threshold and fix #35