arkypita/LaserGRBL #36: G92x0y0 command shifts the work position pointer

witch33, 2017-05-12:
Hi,
After making an offset G92x0y0  the work position pointer moves out and don't follow the workpath. Also may be useful to highlight the work position pointer by colored spot

arkypita, 2017-05-13:
This is not a bug of LaserGRBL since LaserGRBL cannot know where the offset was set, unless it keep count of each G92 in an internal buffer, but this method could be not effective because you can send G92 then exit LaserGRBL, then start LaserGRBL again and in this case LaserGRBL does not know anything about your zeroing.

Position pointer is passivly controlled by the information sent back from grbl in reply of "?" command. This is called "status report".

If you use to set offset prior to begin your job, you may try to configure status report to send 
working position instead of global position. This could be done by setting $10=0 instead of $10=1

witch33, 2017-05-13:
Thank you for the fast answer. Now I understood my problem and tomorrow will try $10=0 setting. I like your program and from my point of view you created the best laser engraving interface to GRBL. 
Best regards

closed 2017-05-13 by witch33

arkypita, 2017-11-07:
[Version v2.8.12](https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.12) can now show working position if different from machine position. Also support $10=0 and $10=1