arkypita/LaserGRBL #37: Question; To

midibase, 2017-05-15:
Hello arkypita,
first off all, let me say you have made a really,really good programm. Your programmstyle is also very readable. :) !!

My understanding is that with the S-command you can tell the laser how intensity it should be.
(Range S0 til S255)
How is it possible to adjust this s-Command on the User-Interface in dependence from the material. 

Explanation:
Before I start to engrave the shape on the fix wood, I would like test it before on a carton. (It absorbes the laser different against wood)
Or when I do a test-engrave on the finished wood, and i think, all shapes are good, but it is to dark so I whis to have it lightener (mean to change the S-Commant) 

So my question is: Is there any possiblity to adjust only the S (Command) value, to get a darker or lighten shape?

Thanks for your opinion.
And best greatings from austria :)

arkypita, 2017-05-15:
Thank you for appreciation
Unfortunately my programming style is far from being really readable, but I do the best I can.
Maybe sooner or later I will do a nice refactoring of the code.

If your hardware use grbl libraryes in version 1.1 you can use overrides feature to easly runtime change laser speed and power.

LaserGRBL offer an easy interface to overrides: http://lasergrbl.com/usage/overrides/

volsan, 2017-05-24:
Good afternoon. On what contacts it is necessary to connect the TTL laser in the CNC shild?

dungjk, 2017-05-24:
@volsan Are you mean connect CNC shield to laser TTL module ? Which type of TTL module you're using (image please) ?

JedOiKnight, 2017-05-24:
Read this to start with >>> https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode

If I recall correctly, when you switch to Laser mode you connect the Laser to Ground and the pin for Z axis End Stop and the Z-Stop switches to a different pin on the Arduino.

There is something written in the Pre-compiled configuration files for Grbl 1.1 ... I'll see if I can find it again and copy/paste it here in a little while ...

JedOiKnight, 2017-05-24:
Read this to start with >>> https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode

I believe the PWM pin is the one on the Arduino CNC Shield board marked as Z+ in the " End Stops " section. I recommend checking PWM output ( variable from 0v to 5v ) on that pin with a multimeter before risking connecting anything up to it.

It would definitely be worth while opening the " config h " file from the Grbl 1.1 source code with a suitable editor such as Notepad++ and having a read through as there is potentially some quite important info in the //comments.

I myself have a 5,500mW laser module but it's running on a Woodpecker board at the moment and is in the process of being switched to Arduino.

Here is a relevant section from " config h " relating to Spindle Speed / PWM which is how you vary the dynamic laser power.

**************

// Enables variable spindle output voltage for different RPM values. On the Arduino Uno, the spindle
// enable pin will output 5V for maximum RPM with 256 intermediate levels and 0V when disabled.
// NOTE: IMPORTANT for Arduino Unos! When enabled, the Z-limit pin D11 and spindle enable pin D12 switch!
// The hardware PWM output on pin D11 is required for variable spindle output voltages.
#define VARIABLE_SPINDLE // Default enabled. Comment to disable.

// Used by variable spindle output only. This forces the PWM output to a minimum duty cycle when enabled.
// The PWM pin will still read 0V when the spindle is disabled. Most users will not need this option, but
// it may be useful in certain scenarios. This minimum PWM settings coincides with the spindle rpm minimum
// setting, like rpm max to max PWM. This is handy if you need a larger voltage difference between 0V disabled
// and the voltage set by the minimum PWM for minimum rpm. This difference is 0.02V per PWM value. So, when
// minimum PWM is at 1, only 0.02 volts separate enabled and disabled. At PWM 5, this would be 0.1V. Keep
// in mind that you will begin to lose PWM resolution with increased minimum PWM values, since you have less
// and less range over the total 255 PWM levels to signal different spindle speeds.
// NOTE: Compute duty cycle at the minimum PWM by this equation: (% duty cycle)=(SPINDLE_PWM_MIN_VALUE/255)*100
// #define SPINDLE_PWM_MIN_VALUE 5 // Default disabled. Uncomment to enable. Must be greater than zero. Integer (1-255).

// By default on a 328p(Uno), Grbl combines the variable spindle PWM and the enable into one pin to help
// preserve I/O pins. For certain setups, these may need to be separate pins. This configure option uses
// the spindle direction pin(D13) as a separate spindle enable pin along with spindle speed PWM on pin D11.
// NOTE: This configure option only works with VARIABLE_SPINDLE enabled and a 328p processor (Uno).
// NOTE: Without a direction pin, M4 will not have a pin output to indicate a difference with M3. 
// NOTE: BEWARE! The Arduino bootloader toggles the D13 pin when it powers up. If you flash Grbl with
// a programmer (you can use a spare Arduino as "Arduino as ISP". Search the web on how to wire this.),
// this D13 LED toggling should go away. We haven't tested this though. Please report how it goes!
// #define USE_SPINDLE_DIR_AS_ENABLE_PIN // Default disabled. Uncomment to enable.

// Alters the behavior of the spindle enable pin with the USE_SPINDLE_DIR_AS_ENABLE_PIN option . By default,
// Grbl will not disable the enable pin if spindle speed is zero and M3/4 is active, but still sets the PWM 
// output to zero. This allows the users to know if the spindle is active and use it as an additional control
// input. However, in some use cases, user may want the enable pin to disable with a zero spindle speed and 
// re-enable when spindle speed is greater than zero. This option does that.
// NOTE: Requires USE_SPINDLE_DIR_AS_ENABLE_PIN to be enabled.
// #define SPINDLE_ENABLE_OFF_WITH_ZERO_SPEED // Default disabled. Uncomment to enable.

**********

Hope this helps ;)

JedOiKnight, 2017-05-25:
Confirmed ... Z+ is the PWM variable voltage output of the CNC Shield BUT ... the Arduino only outputs a very low current and so a suitable Driver circuit / Module MUST be used between the Arduino / CNC Shield and the load in order to adequately drive Motors or Laser modules ... 

[image]

closed 2017-06-05 by arkypita