arkypita/LaserGRBL #39: I do not understand why relative koordinates do work and abslute do not work-crazzy

midibase, 2017-05-17:
Hello 
I use grbl 1.1e

Today I tryed to cut a paper as in the demo-video. This mean that I use "Vectorize!" -Mode.
The result is a g-code in absolute coordinates.
When I send it to eleksmaner SE it makes incorrect positions with terrible noises.
When I try other figure with "Line to Line-Tracing"-Mode I see the g-code is written in relative coordinates. And it do its work perfect.

So my question is: Is there any possibility to say ("G-code please in relative coordinates"?) or do anybody know ´what could be the risen of this crazy behavier when it should move absolute?

Thanks for your help

arkypita, 2017-05-17:
I don't know why eleksmaner SE seem to have problems with absolute coordinates.
Noises usually are related to too high speed for driver/motor.

I am sorry, my line tracing alghoritm produce in relative coordinates, and vectors produce absolute coord, and could not be changed easly.

midibase, 2017-05-17:
thank you.
I have been writing my own programm where I calculate relative corrdinates.
Have a look if it solves my problem
bg

midibase, 2017-05-17:
ok now I test my programm. the transformation works well but do not solve my problem :(

Sometimes movement works well and sometimes it is crazy. 
--> Ok, now I know that it is no softwareproblem. Sorry for my interrupiton

bg

closed 2017-05-18 by arkypita

arkypita, 2017-06-01:
found a solution? what was the problem?

midibase, 2017-06-01:
Thanks for your question:

It seen this is a problem from my Eleksmaker. It has two motors for the y Axis. Sometimes by the movement it makes terrible noices. When I deconnect one side it moves very smoothly. So I think that is a hardware-problem and not a software problem. 
FYI: I set all rules in the correct way and in the right angle, so it seens it is a motor, controller or cable problem. But it moves now with one motor and do its work also ok.