arkypita/LaserGRBL #40: Whats the best way to set the home position?

midibase, 2017-05-19:
Hello,
I would like to set the home position, but who do I that?

My solution is now:
I add a button with "G92X0Y0" to set the home position. The lasermaschine set it right, but
the graphicUI do not set it- This mean the cross in the graphic stand still on the last position.
Homing button doesn't work (($H setting disabled)

So is the any solution?
bg fritz (from Austria)

arkypita, 2017-05-19:
There is a closed issue about this.
https://github.com/arkypita/LaserGRBL/issues?q=is%3Aissue+is%3Aclosed

midibase, 2017-05-19:
Thanks. It helped

lg

 

https://github.com/arkypita/LaserGRBL/issues?q=is%3Aissue+is%3Aclosed

closed 2017-05-19 by arkypita

arkypita, 2017-11-07:
[Version v2.8.12](https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.12) can now show working position if different from machine position.