arkypita/LaserGRBL #41: Image Size

Bricbasher, 2017-05-21:
First off let me say, awesome program.  I really appreciate the time you have taken to do this.

My question is how is image size determined when you enter it in to the Target Image dialog box?  For example I have an image that is 127mm wide by 69mm high.  When I import it and put in the size as 127mm wide and 69mm high it engraves something much much bigger than those parameters.  What am I doing wrong?

Thanks again!!

arkypita, 2017-05-21:
Hi @Bricbasher 
No image size issues are known, are you sure your grbl configuration is correct?
Specifically, $100, $101 values ​​are correctly set?

If you manually send G0 X10 does the laser move exactly of 1cm along X axis?  And the same for G0 Y10, Y axes?

arkypita, 2017-05-21:
note: i mean grbl configuration that is the configuration of arduino board (not LaserGRBL configuration)

you can read grbl configuration by sending $$ command
you can set configuration by sending $number=newvalue

a proper setup of grbl is mandatory to obtain perfect results

Bricbasher, 2017-05-21:
Thank you so much for your reply.  That's exactly what I had wrong.  I knew it was something I was missing.  Please forgive me I'm just learning the ins and outs of my machine and grbl.

closed 2017-05-21 by Bricbasher