arkypita/LaserGRBL #42: Adding support for GRBL plotter

jonpinchen, 2017-05-25:
Nice job, arkypita. 

I wonder if you can add support for grbl plotter.  If you can have an option for free gcode statements on the second screen when importing a picture.  Instead of selecting M3 for laser on, allow user to add gcode 'G0 Z-1.000' for example.  This way GRBL plotter will be able to take advantage of your great work.

Jon

arkypita, 2017-05-25:
nice suggestion!
pen up-pen down :-)

Older versions of LaserGRBL has free textbox for Laser on/Laser off options but newer contains list.
I can made the list editable, so user can add its own options for laser on/off

jonpinchen, 2017-05-25:
arkypita,

That will be great.  Another question I have is why I cannot fill an object with lines.  For example, I imported a solid circle and selected Filling with Horizontal lines.  It showed the circle filling with lines on the preview.  But when I clicked Create, it only showed profile of the circle.  Is there anything I selected wrong?

Thanks.

Jon

[image]

[image]

arkypita, 2017-05-26:
Can you show me the second window (image size and offset) setup?

jonpinchen, 2017-05-26:
Here you go.  Thanks.

[image]

arkypita, 2017-05-26:
This does not happend to me.
While the preview is only a "fake" simuation of the final image not based on gcode data, the image preview is based on an analysis of the gcode data.

First question: the problem is only that preview did not show horizzontal filling lines, or the problem is that the lines was not generated? In other words, if you execute the program the filling will be done or not?

Determine when the laser is on, and when the laser is off, and how to make gray tones according to the S0-S255 range is a tedious job. I.e You can have M3 active but if you have S0, the laser is off.

To understand what's happening on your LaserGRBL instance I ask you to also send me the settings file, located in: C:\Program Files (x86)\LaserGRBL\LaserGRBL.Settings.bin and the session log located in C:\Program Files (x86)\LaserGRBL\sessionlog.txt

And also a copy of example g-code exported with "file -> save program" whould be appreciated.

None of them contains sensitive information, you can zip them and upload here.

jonpinchen, 2017-05-26:
I found out the problem occurred when I unchecked Support PWM in the settings dialog.  The reason I unchecked it is because when PWM is enabled, the gcode file it generated uses S codes to draw the lines by appending S code in the end of each line.  

When it is disabled, it generates a separate M3 or M5 line for each laser on or off.  I was globally replacing M3 and M5 to different codes. That is the reason I asked you for adding the feature for plotter.

I have attached all the files.  

somehow I could not drop a zip file.  So I renamed them all to .txt.  Hope it works.

Thanks.

-Jon

[circle_solid_gcode.txt](https://github.com/arkypita/LaserGRBL/files/1032173/circle_solid_gcode.txt)
[LaserGRBL.settings.txt](https://github.com/arkypita/LaserGRBL/files/1032175/LaserGRBL.settings.txt)
[sessionlog.txt](https://github.com/arkypita/LaserGRBL/files/1032174/sessionlog.txt)

arkypita, 2017-05-30:
i have found where the problem is, but i have not time now to fix it
maybe next week

arkypita, 2017-06-06:
The reason why filling does not work for you, is because you set-up laser min power = 10 and laser max power = 10 in target image form. Set the same value for min max cause line filling algorithm to see this color as white, also if used on a "non pwm" engraver.

I have put 2 fix in code:
- prevent setting of invalid power range in this form
- ignore power range when in "non pwm" mode

arkypita, 2017-06-06:
I will soon add the ability to set pen-up / pen-down and then I will release a new version.

closed 2017-06-06 by arkypita

arkypita, 2017-06-06:
dear @jonpinchen 

I've tried adding pen-down per-up functionality, but it's not as simple as I thought.

The Z codes can not be inserted simply in place of M3/M5 because while the M3/M5 can be inserted in line with the rest of the command (i.e., G1 X10 M3) the Z codes need an autonomous execution line, especially if they are to be run with a specific G mode.

Thus making LaserGRBL fully compatible with this mode requires the rewriting of some GCode generation, optimization and drawing routines.

jonpinchen, 2017-06-06:
arkypita,

I understand that it is not as simple as just replacing the M3/M5 S0/S10
with G1 Z? statements.  You need also consider which mode the printer is
in, ie. G90 or G91.
I wrote small add-in with VB script in Excel to convert GCode file
generated by LaserGRBL.  It changes to G90 (absolute mode) whenever it
encounters G91 and reverses back to G91 after the G1 Z? statements.   It
works pretty well so far.  Here is part of the code.

Appreciate your help on this.  If possible, I still hope you can have the
program be compatible with plotter.  It is a really neat program.

-Jon

  bG91 = False
  sPenDown = "G1 Z-2 F3000"
  sPenUp = "G1 Z7 F3000"

  On Error Resume Next

  Open sInputName For Input As #1
  Open sOutputName For Output As #2

  Do Until EOF(1)
    Line Input #1, sBuf
    If Left(sBuf, 2) = "M3" Then
      If bG91 Then Print #2, "G90"
      Print #2, sPenDown
      If bG91 Then Print #2, "G91"
    ElseIf Left(sBuf, 2) = "M5" Then
      If bG91 Then Print #2, "G90"
      Print #2, sPenUp
      If bG91 Then Print #2, "G91"
    ElseIf Right(sBuf, 3) = "S10" Then
      If bG91 Then Print #2, "G90"
      Print #2, sPenDown
      If bG91 Then Print #2, "G91"
      sBuf = Left(sBuf, Len(sBuf) - 3)
      Print #2, sBuf
    ElseIf Right(sBuf, 2) = "S0" Then
      If bG91 Then Print #2, "G90"
      Print #2, sPenUp
      If bG91 Then Print #2, "G91"
      sBuf = Left(sBuf, Len(sBuf) - 2)
      Print #2, sBuf
    ElseIf Left(sBuf, 3) = "G91" Then
      bG91 = True
      Print #2, sBuf
    ElseIf Left(sBuf, 3) = "G90" Then
      bG91 = False
      Print #2, sBuf
    Else
      Print #2, sBuf
    End If
  Loop

arkypita, 2017-06-07:
> You need also consider which mode the printer is
in, ie. G90 or G91.

Yes, you are right.
LaserGRBL use both G90 and G90. Line to line filling is easyer to generate using relative movement when vectorization alghoritm come out with absolute coords so it is easyer to create G-Code with absolute coord.

Sanyok-1, 2017-06-26:
Hello!
Very much like your program, especially what can draw with filling!
Very much waiting for the plotter mode.
Thank you for your work.

arkypita, 2017-06-27:
be patient, it will come soon (as soon as I have some free time and it's raining!)

jonpinchen, 2017-06-27:
It is good to know more people are interested in plotter mode.  I am sure this program will make life easier.  Can't wait to see the new feature.  Thanks.

slanwar, 2018-05-08:
I use this program to plot, I'm using a servo and works with the commands M3/M5, great program because we don't need Inkscape to create gcode files.

antoniocanabarro, 2018-09-26:
finally I discover a firmware to move a servo with a pen and drawing in lasergbrl like a charm

arkypita, 2018-09-26:
> finally I discover a firmware to move a servo with a pen and drawing in lasergbrl like a charm

link?

antoniocanabarro, 2018-09-26:
im using a chinese arduino uno and shield v3 with a4988 drivers the trick is in windows notepad replace  m3 to s80 is the servo angle and m5 to s5 (servo angle to) the firmware version is 1.1  in archive has a sample file https://drive.google.com/drive/folders/1z4G5Ep34IYkXk4ehFAY4hI5gonAe8vog?usp=sharing

antoniocanabarro, 2018-09-26:
the file is generate in lasergrbl, this is the best because no need inkscape to translate the gcode

antoniocanabarro, 2018-09-26:
https://youtu.be/Y2nZ3xZmox8

garthanstey, 2019-06-29:
I have just converted to a grbl plotter where M3S0 - M3S255 controls servo motion. I can't get control of it with Laser grbl  or set a delay to allow for pen up/down delay. Is there a version of grbl that does that?

commit 5e442aa3 (2017-06-06): fix #42