arkypita/LaserGRBL #43: Rapid moves shown as cuts

Ranthalion, 2017-05-28:
G0 rapid moves are shown as cuts when loading gcode created by other CAM software.  

[image]

[enclosureUpdate.Part1.zip](https://github.com/arkypita/LaserGRBL/files/1034218/enclosureUpdate.Part1.zip)

arkypita, 2017-05-28:
For what I can see the behaviour of LaserGRBL it is perfectly reasonable and aligned to what the engraver will do.

If I look at the first lines of the code I can see:

G90 G21
G64 G17
M3 S1000       //power on laser @ power of 1000
G0 Z3.0           //rapid move
G0 X33.3564 Y17.611      //rapid move
G0 Z1.4     //rapid move
G1 F300.0 Z0.0   //normal move - Z0.0 what???
G1 F400.0 Y24.741
G1 X133.1364

arkypita, 2017-05-28:
If laser is switched on it will burn the material, no matter of G0, G1, or Z command

The problem is in the cam program you had used, if you use a cam that is designed for lathe or cnc it will suppose that the spindle is turned on at start and never turned off, only raised and lowered to engrave. This could be ok for CNC but laser are different

Ranthalion, 2017-05-28:
According to the Grbl v1.1 docs, the laser is always turned off during G0 rapid movement. 
from https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode
"The laser will only turn on when Grbl is in a G1, G2, or G3 motion mode.

In other words, a G0 rapid motion mode or G38.x probe cycle will never turn on and always disable the laser, but will still update the running modal state. When changed to a G1 G2 G3 modal state, Grbl will immediately enable the laser based on the current running state."

I am new to using Grbl, so it may be that this is not the usual expected laser behavior in older versions.  I confirmed that grbl did not cut during the rapid movements with G0 when cutting this piece.  

The Z movements don't impact the operations in this case, the gcode was generated using Cambam - which is primarily intected for CNC milling.

Thanks for all the effort on LaserGRBL.  I've really enjoyed how simple it makes raster engraving!

Ranthalion, 2017-05-29:
I added a dotted blue line to represent rapid moves when the laser may be powered on; it does not show the rapid moves when the laser power is off according to the gcode.  It also isn't configurable, but if you approve of the change and desire, I can move it to a configuration setting.  I'll submit a pull request for your review shortly.

[image]

arkypita, 2017-05-29:
It's ok for me if you submit a pull request, but for a few days I will not be able to work on it, but I will look at it as soon as possible when I am back.

LaserGRBL claims to be compatible with GRBL v0.9 and v1.1 but the "LaserMode" feature is new to v1.1 so the preview should know in advance if the version support "LaserMode" and whether the Laser Mode is active or not.

This is not so easy because it is always possible to load the file even when the marker is disconnected, so without the ability to query the grbl version and its capabilities and configurations.

From this it follows that to draw, it is necessary to make an assumption: G0 writes or G0 does not write. Anyway there is always someone who will complain that the design is different from reality.

arkypita, 2017-05-29:
Configuration is the right way. Go on it!

Ranthalion, 2017-05-29:
Got it.  I'll add the configurations setting and send an update in a couple of days.

closed 2017-06-06 by arkypita