arkypita/LaserGRBL #45: can't get right gcode

AndySze, 2017-05-30:
Hi, @arkypita , I'm using LaserGRBL to generate the Gcode for my laser engraver.
but, when I'm trying to use Filling or Line to Line tracing, the Gcode will be wrong.

Vectorize No Filling is working:

> G90
F4000
G0 X0 Y0
M5 S255
G1 F1000
G0 X63.114 Y51.703
M3
G1 X62.75 Y51.706
G1 X62.55 Y51.817
G1 X62.35 Y51.927
G1 X62.119 Y52.136
G1 X61.889 Y52.346
G1 X61.591 Y52.865
G1 X61.294 Y53.385
G1 X60.848 Y53.881
G1 X60.402 Y54.377
G1 X59.408 Y54.801
G1 X58.414 Y55.226
G1 X58.082 Y55.569
G1 X57.75 Y55.913
G1 X57.759 Y56.753
G1 X57.769 Y57.592
G1 X58.079 Y58.221
G1 X58.389 Y58.85
G1 X58.942 Y59.519
G1 X59.494 Y60.188
G1 X60.192 Y60.602
G1 X60.889 Y61.016
G1 X61.36 Y61.158
G1 X61.832 Y61.3
...

with Filling option or Line to Line will generate gcodes like this:

> G90
F4000
G0 X0 Y0
M3 S0
G1 F1000
G91
G1 X49.667 Y10.989 F4000 S0
G1 F1000 X-3 S1
G1 X-1.333 Y0 F4000 S0
G1 F1000 X-1 S1
G1 X-1 Y0.333 F4000 S0
G1 F1000 X.333 S1
**X.333 S23
X.333 S70
X.333 S123
X.333 S169
X.667 S185
X.333 S184
X.333 S151
X.333 S144
X.333 S146
X.333 S145**
X1.667 S143
X.333 S144
X.333 S140
X.333 S57
X.333 S1
G1 X0.334 Y0.333 F4000 S0
G1 F1000 X-.333 S4
X-.333 S154
X-6 S255
X-.333 S232
X-.333 S153
X-.333 S69
X-.333 S20
X-.333 S1
G1 X-0.333 Y0.333 F4000 S0
G1 F1000 X.333 S25
X.333 S138
...

very wired, don't know why.
Hope you can solve this problem. Thanks.
Andy.

arkypita, 2017-05-30:
can you provide me a sample image file on which you are experiencing problems, and your configuration file so I can reproduce the issue on my PC?

your configuration is stored in a settings file, located in: C:\Program Files (x86)\LaserGRBL\LaserGRBL.Settings.bin

AndySze, 2017-05-31:
sure,

[image]

the config file: https://drive.google.com/file/d/0ByK3W2c482S4eXlTVm8yX1FFUWM/view?usp=sharing

arkypita, 2017-06-01:
i am out of home this week, but I will look at it when i can

arkypita, 2017-06-06:
Does your engraver support PWM power modulation?

As expressly highlighted on the homepage "In order to use all of LaserGRBL feature, your engraver must supports laser power modulation through gcode "S" command."

AndySze, 2017-06-06:
Yes, my machine support PWM. My problem is the X.333 parts.
you can see, lots of X.333 code. it's unusual.

arkypita, 2017-06-06:
It is not inusual, it is how laserGRBL generate lines.
I suppose you have 3lines/mm true? Mean that your line and your pixel are 1/3mm=0.333mm

Because of the [G91 command](https://www.tormach.com/g90_g91.html), all XYZ movement are to be assumed as relative movement, not absolute coordinate

So this code say:

```
X.333 S144 --move 1px color (power) 114
X.333 S146 --move 1px color 146
X.333 S145 --move 1px color 145
X1.667 S143 --move 5px color 143...
```

arkypita, 2017-06-06:
ps. the image will be smoothed out at the border to obtain not jagged border because of the resize/rescale alghoritm, so a lot of X.333 is due the effect of this smoothing

if you dont' like smoothing you can apply b/w threshold to your image

closed 2017-06-06 by arkypita

fesch, 2017-12-20:
@AndySze : I've compiled a version that generates GCode accepted by Repetier firmware. Interested?

iwasbeingsarcastic, 2019-03-15:
> 
> 
> @AndySze : I've compiled a version that generates GCode accepted by Repetier firmware. Interested?

Id be interested. Ive figured out that the gcode generated is missing the "s" parameter on G1 moves. Id like to do vector images, but i have to manually add S255 on the G1/2/3 moves. Also, any travel moves not preceded by a G1 command is ignored by my firmware.  Ive been looking for a solution.