arkypita/LaserGRBL #46: multiple issues - laseraxe 5500mw laser

wesleycwalker, 2017-06-01:
Hello first off I love the way your software looks,  I truly wished it would run with my engraver.  Please advise if there are tricks to make it run without having to update firmware.

1. I've tried line, dither, vectorize - the engraver would move to the first position, but never engage or move from that point.

2. In creating nc files I able to export them in input them into my cncc program the vectorized nc files work great. the line and dither when importing them show like a line going from left to right.  I would love to get the gray scale with either dither or line to work any suggestions?

Please advise on any idea's I can try to get your software to work with my engraver or suggestions to get the exported nc files to work.  I've also exported from your software a working vectorize file versus a non working dither file.   uploading to this link change the .txt to .nc for the correct file type.

Sincerely,
Wesley Walker

[cheyenne test - .txt](https://github.com/arkypita/LaserGRBL/files/1046003/cheyenne.test.-.txt)
[humming bird test - .txt](https://github.com/arkypita/LaserGRBL/files/1046004/humming.bird.test.-.txt)

wesleycwalker, 2017-06-01:
updated I was looking over the code for Cheyenne test - dither test - found that the Y never changed the software is software is only showing Y0.  Just a FYI not sure if this is a bug.

G90
F4000
G0 X0 Y0
M3 S0
G1 F1000
G91
G1 X17.667 Y0.334 F4000 S0
G1 F1000 X.167 S255
G1 X3.167 Y0 F4000 S0
G1 F1000 X.167 S255
G1 X1.667 Y0 F4000 S0
G1 F1000 X.167 S255
G1 X1.167 Y0 F4000 S0
G1 F1000 X.167 S255
G1 X10.167 Y0 F4000 S0

but this explains why it shows it in 1 line.

arkypita, 2017-06-01:
i am out of home this week, but I will look at it when i can

AndySze, 2017-06-02:
it's the seam issue with #45 , I think.

wesleycwalker, 2017-06-02:
I can't tell was there a fix for this problem?  Do you need my config file and a picture?

arkypita, 2017-06-03:
Could be usefull.
A solution is not ready because i cannot work on it im this days

arkypita, 2017-06-06:
it is not the same issue of #45

arkypita, 2017-06-09:
If I understand correctly laseraxe uses a different firmware that is not standard-compatible with Grbl.
LaserGRBL is only compatible with GRBL, so simply does not work with laseraxe firmware.

closed 2017-06-22 by arkypita