arkypita/LaserGRBL #49: problem mit laser on off

goblin84, 2017-06-04:
Habe ein problem mir laser on off also M03 M05 meiner geht an (on) bei M04 und aus (off) M03
Wie kann ich es ändern das er bei M03 an geht und bei M05 aus bitte um hilfe

arkypita, 2017-06-05:
The question is not clear, I tried translating it with "google translate" and it is even less understandable.
Please write your question in english

arkypita, 2017-06-05:
If I understand well, your laser works in reverse: turned off with M3 and turned on with M5. Is that so?

goblin84, 2017-06-05:
Yes almost M4 on  M3 off and M5 no function

My grbl is standart 1.1e

arkypita, 2017-06-06:
If your laser is off on M3/M4 maybe it is a problem with your hardware, not with LaserGRBL. So why ask here?

Please note that with GRBL v1.1 and "Laser Mode" laser turn on only on G1,G2,G3 movements and automatically shut down if no movement in progress or in G0 fast movement.

closed 2017-06-07 by arkypita