arkypita/LaserGRBL #53: Versetzte Füllung

herbiro, 2017-06-15:
Hallo. Wenn ich etwas mit dem Laser mache verliert er mittendrin einfach Mal die Position und versaut mir das ganze. Ausserdem ist mir aufgefallen das seit dem Update die Füllung versetzt gelästert wird. How to fix it?

arkypita, 2017-06-16:
The request not clear. I don't know german and google translator did not help.
It would be better if you write in English.

If I interpretate correctly your question:
- in previous version v2.6.8 all is fine
- in current version v2.6.9 you have problem

Before version 2.6.9 all the moves generated by raster import were made with G1 (controlled linear movement). In v2.6.9 i have changed the use of G1 with the use of G0 (rapid linear movement) for non tracing movement when the laser is off.

Maybe, the speed you set-up for rapid movement in raster import image size and offset is too high for your hardware? 
[image]

Did you have a proper configuration of grbl acceleration?
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#120-121-122--xyz-acceleration-mmsec2

Here a test you can do to check if the problem is related to G1-G0 changes:
- generate image with raster import
- save the generated G-Code with "File - Save"
- open the generated file with notepad 
- replace all G0 with G1
- open the modified file and engrave it

If the result is corrct the problem is in how your hardware execute G0

Toiletfly, 2017-06-16:
Hello Arkypita,

I guess I have the same problem as herbiro and I'm from germany too, so I'll try to explain.
When I'm using high engraving speeds (4000mm/min and above) in combination with a high filling quality (more then 20 lines/mm), then GRBL looses the stepper motor position and the engraving is continued a few mm shifted in the filling direction (horizontal / vertical /...).

When the filling quality is set to 10 lines/mm and less, everything is fine ever with higher engraving speeds.

Also if I set engraving speed to 4000mm/min or higher und filling to 20 lines/mm und use a really small picture to engrave, (aprox. 15x5mm) everything runs fine too.

When I look into the G-Code the prgramm has created, I see it switches to G91 at the beginning of the programm (incremental moves) and with 4000mm/min and 20 lines/mm it generates a lot of moves for the axis. My guess is that GRBL doesn't get along with that much code in such short time.

For example:

G91
G0 X4.85 Y53.65 F4000 S0
G1 F4000 Y-.05 S61
Y-.05 S76
Y-.1 S90
Y-.05 S76
Y-.05 S57
G0 X0.05 Y-0.35 F4000 S0
G1 F4000 Y.25 S53
Y.05 S57
Y.05 S94
Y.05 S139
Y.05 S195
Y.05 S236
Y.05 S232
Y.05 S169
Y.05 S90
Y.1 S53
G0 X0.05 Y0.1 F4000 S0
G1 F4000 Y-.05 S61
Y-.05 S105
Y-.05 S184
Y-.05 S288
Y-.05 S392
Y-.05 S463
Y-.05 S504
Y-.05 S500
Y-.05 S467
Y-.05 S426
Y-.05 S381
Y-.05 S348
Y-.05 S292
Y-.05 S232
Y-.05 S191
Y-.05 S161
Y-.05 S143
Y-.05 S113
Y-.05 S83
Y-.05 S61

 and so on-......

This describes a peace of a dot which has a diameter of about 4mm. These are alot of move orders.
I selected "Black and white" in hope to reduce the amount of lines, but it didn't work. I expected behind every move order would stand a "S1000" because I turned grayshades of.

The Programm would be more efficient if the lines above would just be (theoreticly it would be the same):

G91
G0 X4.85 Y53.65 F4000 S1000
G1 Y-3 F4000 S1000
G0 X0.05 Y-0.35 F4000 S0
G1 F4000 Y.525 S1000
G0 X0.05 Y0.1 F4000 S0
G1 F4000 Y-10 S1000

If I havn't made a mistake the Laser would do exactly the same as in the programm snipet above, only with less G-Code Lines to process (In Black and White I wouldn't need less laser power than 100%. So Grbl wouldn't be overloaded.

Do you have any idea how to solve this?
I hope my english isn't that bad and you understand what I am trying to say.

Regards

arkypita, 2017-06-21:
Image scaling (enlargement, reduction) cause smooth image edges, smooth edges are reproduced by small step at different laser power.

[image]

`I selected "Black and white" in hope to reduce the amount of lines, but it didn't work. I expected behind every move order would stand a "S1000" because I turned grayshades of.`

Sound strange. I have tested it just now, and this is my output:

```
G91
G0 X39.05 Y2.85 F4000 S0
G1 F1000 X-1.95 S255
G0 X-0.85 Y0.05 F4000 S0
G1 F1000 X3.6 S255
G0 X0.65 Y0.05 F4000 S0
G1 F1000 X-4.9 S255
G0 X-0.65 Y0.05 F4000 S0
G1 F1000 X6.2 S255
G0 X0.45 Y0.05 F4000 S0
G1 F1000 X-7.05 S255
G0 X-0.35 Y0.05 F4000 S0
[...]
```

rucncmasterkit, 2017-06-21:
I think the problems are caused by a lack of computer RAM. I checked the performance of my equipment and found out that a picture of 20x20 cm in size with 10 lines per mm uses 1500 megabytes of memory due to which a crash occurs.

closed 2017-06-29 by arkypita

mdeel-online, 2020-04-16:
Is it possible to reactivate the old commands in Lasergrbl version 3.2.0 without having to use the notepad every time?
[image]

arkypita, 2020-04-16:
Wouldn't it be better for you to fix your engraver configuration to solve the problems it has with G0?
G0 is the standard command for fast positioning, and your machine definitely supports it.

Fast positioning use the max speed set in configuration.
If you have problems with fast positioning because your machine is inaccurate when positioning, you must lower the acceleration and maximum speed values .

[image]

mdeel-online, 2020-04-16:
Thank you for the answer, but can you tell me what to put in there?

arkypita, 2020-04-16:
$120 - $121
When the motors move the axes they must accelerate from zero speed - since at the beginning they are stopped - up to the required speed.

In the same way when there is a change of direction the motors must slow down - almost to a stop - and then accelerate again in the new direction.

If the acceleration is too high the weight and inertia of the mechanics make the engine fails positioning, or the belts slip and lose steps. If steps are lost, the position is inaccurate and there is no way that the software can notice, because there is no feedback from the motors and the belt.

$110 - $111
There is a maximum speed that your marker can support. It depends on the motors it mounts, on the quality of the control electronics, on the mechanics, on how the belts are regulated.
If you perform too fast a movement, and your mechanics cannot do it, the motors simply won't move - or they will jerk, or they will slip.

**Warning**: all G0 fast movements are performed at this speed. It is therefore very important that the configured value is precisely managed by the machine.

**The values to be configured vary from machine to machine, so you have to find them experimentally.** 

I think I will make a video to help people find the best values for their engraver.

mdeel-online, 2020-04-16:
Thank you for the detailed answer. I think I will continue with Notepad and then somehow try to find out. I thank you. 
TAKE CARE OF YOU!!!!