arkypita/LaserGRBL #55: The distortion of the image when burning

AlaAndr, 2017-06-16:
Hallo, Arkypita
I have the same problem, as the authors of two previous posts. This happens is, during execution of the code,  the cursor jumps to another location and begins to go on a white background (video for this error is in the post from StuartB4),  and the figure burned with a distortion. This It happened sometimes  in version 2.6.7. After upgrading to 2.6.9 error always occurs and use the program I can't. Toiletfly wrote that he has such error appears on big drawings or when as more than 20 lines per mm. In my case it occurs in all drawings and in all modes. I use a speed of 3000-4000 mm/min. reinstall the firmware GRBL and a return to the older version of the software did not help.
Please,  help me.
I write through the translator, probably, not everything turned out clear. I attached the image with  my experiments
[https://yadi.sk/i/drmpab2k3KCfUP](url)

arkypita, 2017-06-16:
Can you send me the gcode?
I will try with my engraver on sunday afternoon

AlaAndr, 2017-06-17:
Hi. Today I spent some more tests.
 I have created and saved the code in LaserGRBL.  Then, this code I did in the Candle program.  In the "Line to Line tracing" appears the error message in those places, where then there has been a shift of engraving. I made a screenshot of these errors. Maybe this will help solve the problem.
 In the "Vectjrize" message was not, but a slight offset is noticeable.
[error.pdf](https://github.com/arkypita/LaserGRBL/files/1082720/error.pdf)
[https://drive.google.com/file/d/0B563a_78_Ew0emJvSkx1bm92cVE/view?usp=sharing](https://drive.google.com/file/d/0B563a_78_Ew0emJvSkx1bm92cVE/view?usp=sharing)
[https://drive.google.com/file/d/0B563a_78_Ew0MklWSUgxR3UyNTg/view?usp=sharing](https://drive.google.com/file/d/0B563a_78_Ew0MklWSUgxR3UyNTg/view?usp=sharing)

arkypita, 2017-06-18:
I try executing both code with my DIY engraver, without any problem.

[image]
[image]

In my opinion you could have wrong configuration in your grbl setup on arduino side, that made your motor lose step when G0 at fast speed.

Try lowering $120, $121 and/or use lower engraving speed.
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#120-121-122--xyz-acceleration-mmsec2

Also... as you can see in grbl documentation error 25 mean "A G-code word was repeated in the block." 

[image]

but as you can see in your error.pdf file, there are NO REPETITION in this gcode block. So.... where is the problem? The problem is that you have electrical noise on your usb cable, caming probabily from motors. You are not the first, you are not the ones, but it is not related to LaserGRBL.

https://github.com/arkypita/LaserGRBL/issues/24

AlaAndr, 2017-06-19:
Thank you. I also thought it was  is a disconnection. I incorrectly translated  "Jogging speed" -I thought it was the value control   $120 - 121. Changed them in different ways, but a positive result was not. Now changed in GRBL - $120-121=700 (was 3000). The movement is very slow, errors became less, but they happen. I will  to buy a new USB cable. The jump of voltage electrical network can affect its? Maybe even a surge protector to buy?

arkypita, 2017-06-20:
> I will to buy a new USB cable. The jump of voltage electrical network can affect its? Maybe even a surge protector to buy?

It is difficult to answer these questions in a general way. Every marker is different, different in power transformer, different motors, different cables, different drivers, different arduino clones (really few people use the original arduino, which has more reliable components and serial chip).

But luckily the cause is always the same: electrical noise generated by stepper motor. Stepper could generate very strong electrical spike on power line that can go back to USB line.

The cable is the first inquired. In my experience I have tested 3-4 different cable and only one is 100% reliable. The good one is a samsung phone data cable. Good ones should be short, thick, well shielded, and possibly with ferrite. www.google.it/search?q=usb+with+ferrite

Some other suggestion:

- if you use a laptop plug it to the power line and do not run with battery. doing this you have grounding through main power line. 
- place your USB cable far from motors and DC line, to minimize induced noise.
- add a pair of capacitor parallel to DC line, one elettrolitic of 47uF and one ceramic of 0.1uF. this is called "[decoupling](http://www.thebox.myzen.co.uk/Tutorial/De-coupling.html)" and could help in case of electrical noise.

arkypita, 2017-06-21:
tell me when you solve and how. could be helpful for other users

closed 2017-06-29 by arkypita