arkypita/LaserGRBL #56: S12000

CaffeineOverdrive, 2017-06-20:
The GRBL version I am using allows simultaneous 3 Axis CNC spindle control and laser control. The two issues I have are:
1. I cannot jog the Z axis for focussing the laser. 
2. The firmware S command range is 0 - 12000. The option to change the max S command tops out at 255.

arkypita, 2017-06-21:
1) you can send any gcode command you want, via [custom button](http://lasergrbl.com/en/usage/custom-buttons/) or via [manual command](http://lasergrbl.com/en/usage/user-interface/)

2) it is true, I can open to large values in input, but it is really needed? I mean, arduino can only PWM 8 bit, so 255 different power values. Grayscale image are 8 bit, so only 255 different shades of gray. Having a S range of 12000 in firmware is really useless. Why you don't normalize to 0-255 like other people?

arkypita, 2017-06-21:
In [latest version](https://github.com/arkypita/LaserGRBL/releases/latest) I have removed upper limit for S range.

closed 2017-06-22 by arkypita

CaffeineOverdrive, 2017-06-29:
Thanks,
I've already used it. It works great thanks!