arkypita/LaserGRBL #58: New feature

e6910, 2017-06-22:
This SW is just great. All I need so far.

It sould be great, however, in the vectorize option to be able to specify how many passes to perform on the traces.

Also the ability to control how the engraving the work based in colors, like Laserdraw does, I mean, black surfaces will be filled, red lines will be cut and blue lines will be traced

That are my suggestions.

Marino

arkypita, 2017-06-23:
> It sould be great, however, in the vectorize option to be able to specify how many passes to perform on the traces.

Sounds good, but I prefer to control n° of passes in a second step (repeating more time the same single-pass gcode, and not by creating a repeated gcode file). I think it is more flexible approach

Also the ability to control how the engraving the work based in colors, like Laserdraw does, I mean, black surfaces will be filled, red lines will be cut and blue lines will be traced

I don't know LaserDraw, but for what I see the feature you are talking about is implemented in a vectorial drawing editor. Here we have raster-to-laser conversion, so it is not so easy to distinguish what is black, what is red, what is a line and what is a path etc.

When I will do the vector import (i.e. to open file from .svg or .dxf) i will create some kind of association with layers and process (speed, direction...), and between colors and process. But it is a desiderata, I am not planning to develop it in the immediate future.

e6910, 2017-06-23:
Humm I will try that approach, tracing with no filling.

Laserdraw is an extension for inkscape used to produce files to work with chea laser engravers. I didnt think of th fact that we are dealing with bitmaps. As you say, this would be an option once the import vector module is working.

One more question, can I import gcode files made with other softwares? I tried, bu all I got was either a blank screen or random lines and circles.

Any way, yours is the besto tool to work with grbl based cnc (I've tried 3 or 4 so far)

Excellent work!!!

Marino

closed 2017-07-03 by arkypita