arkypita/LaserGRBL #60: The distortion of the image when burning-2

AlaAndr, 2017-06-30:
Hi, I see that you closed my question.
My problem has been solved only partially. Replacement USB cable will not be improved the control of the machine. But I realized that is causing this problem. It happened after I put all the wires in the cable channels. The wires were of different lengths, and some of them were folded in loops.This, obviously, increased electrical noise from the engine and caused disruptions in the transmission of data via the USB cable. After the wire was removed from the cable channel, significant image shifts disappeared. 
There is still another problem.
Wavy distortions of the image continue to occur.
[image]
( left-horizontal, right-vertical)
It appears at the fill mode horizontal and diagonal. When the fill mode is vertical, picture is accurate. For vector there is also a small shift, even during vertical filling.   What it is? Mechanical barriers to movement are absent, the cables do not catch on. Likely that error is caused by a rapid movement along the X-axis?  But why quick move on the Y-axis do not cause such interference. How to fix this error? I would like to use diagonal mode and vector.

arkypita, 2017-06-30:
1) This problem happens at any speed or only at high speeds?
If it happens only at high speeds: motor step loss or belt slippage or improper belt tension or too low motor power. Please note that usually one axis is heavier than the other axes. One axis motor move only laser head, but the other motor move the laser + the laser rail + the other motor... that is a lot of weight!

If you see at [my design](http://lasergrbl.com/wp-content/uploads/2017/03/DSC_1457.jpg) I put two motors in parallel (lower left and lower right, clipped) to move the laser rail to prevent this problem. 

Solutions:
- Try decreasing max speed and acceleration of your grbl $$ setup
- Check belt tension and stepper driver current
- Put some oil on rail

2) Do you have the same x/y resolution (your motor/gear/belt is the same in x and y) or different resolution? Since LaserGRBL use only one "line/mm" data to generate x/y coords, if you have different resolution in x and y some rounding could be applied by grbl. Those rounding could act differently in x and y and create some kind of aliasing/lines/distortion. This problem exist also if you don't use a reliable resolution.

https://github.com/arkypita/LaserGRBL/issues/30

Solutions:
- If possible, build a same x/y resolution hardware is better then asymmetric
- If you have same xy res, try compute a reliable resolution with "[resolution helper](https://cloud.githubusercontent.com/assets/8782035/25584789/b9ccecf6-2e97-11e7-88dc-9ea01e294006.jpg)" 

so... make this test, try to understand what's going wrong, and tell me

closed 2017-07-14 by arkypita

AlaAndr, 2017-07-14:
I solved this problem by adjusting belt tension.The tension was weak. 
Now the image is distorted relative to the x-axis, sometimes a little.
I will use the "vertical mode" for complete accuracy, and in "vector" mode, I reduce speed to 2,000 mm / m.
Maybe I'm asking stupid questions, but I'm a girl. Therefore, it is necessary to "understand and forgive".
This problem is solved. Thanks for the help.
Your program is the best of all tested by me.

I have another question, I'll create a new topic.

arkypita, 2017-07-14:
You are welcome.