arkypita/LaserGRBL #62: baud rate 250000 velleman k8200

swepeter, 2017-07-06:
Hi,

I am new to this and hope you can help me. 

I am running a velleman k8200 3d printer that I want to use as a laser engraver. 
The baud rate I have defined in the 3d printer software i 250000 and I do not know how to re compile the software to add this. 

Do you belive the velleman k8200 can be used with this application

looks like an awesome software! 

Thanks
/Peter

arkypita, 2017-07-07:
The available speed list is coded in file [ConnectLogForm.cs line 19](https://github.com/arkypita/LaserGRBL/blob/master/LaserGRBL/ConnectLogForm.cs#L19)

You can [download the whole source code](https://github.com/arkypita/LaserGRBL/archive/master.zip) and edit the list. Then compile the source using visual studio or [sharp-develop](http://www.icsharpcode.net/OpenSource/SD/Download/Default.aspx)

arkypita, 2017-07-07:
Does k8200 use Grbl? 
LaserGRBL send and receive standards G-Codes, but some feature (homing, status, position...) are controller-specific, and as the name implies LaserGRBL is designed to be used with grbl controller.

swepeter, 2017-07-07:
Version 1 of the k8200 firmware called Marlin based on Sprinter and grbl
Reprap firmware. Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm

swepeter, 2017-07-07:
Thank you for the quick reply =)

2017-07-07 10:06 GMT+02:00 Peter Andersson <danspeter@gmail.com>:

closed 2017-07-14 by arkypita