arkypita/LaserGRBL #63: GRBL 1.1 no points

UmkaAP, 2017-07-10:
Good afternoon. Prompt where to dig. Firmware GRBL 0.9 small details
[image]
 burns, and 1.1 ignores them.

arkypita, 2017-07-10:
Is this a question about LaserGRBL or about grbl?

LaserGRBL generates and sends exactly the same gcode, regardless of the grbl version with which it communicates.

So if the question regards grbl It would be better to post on [grbl github](https://github.com/gnea/grbl/issues) where more competent people can answer.

However I think that the differences will be due to the M4 laser mode added with v1.1
[Grbl-v1.1-Laser-Mode](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode)

UmkaAP, 2017-07-10:
Thank you. Why is there a difference of $32=1 or $32=0.

arkypita, 2017-07-10:
grbl was originally designed for CNC milling machine, a CNC mills usually have a spindle that rotate at certain speed: spindle speed could be variable and controlled via PWM by S command. A laser can be equated to a spindle with power controlled "like the speed" but with some fundamental differences.

- spindle need time to change it own speed; laser power change is instantaneous
- if axes motion stop with powered spindle nothing bad happens; if motion stop with laser turned on, it continue burning
- if motion accelerate/decelerate to follow angles and direction changes nothing bad happens with a spindle, but with a laser any acceleration-deceleration will result in a darken/lighter mark.

Up to version 0.9 grbl had no specific support for the laser. Who used grbl 0.9 with lasers accepted some limitations due to this lack.

Starting from version 1.1 grbl developers have added this new "laser mode" to grbl that should be enabled with $31. As I have already told you, all documentation and information on grbl laser mode can be found on the grbl site. [Grbl-v1.1-Laser-Mode](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode)

UmkaAP, 2017-07-12:
The problem was the PWM frequency. Version 0.9 uses 8 kilohertz, and in version 1.1 the default is 1 kHz.

arkypita, 2017-07-12:
Great catch! I hope you know that you can change it in [file cpu_map.h
](https://github.com/gnea/grbl/blob/master/grbl/cpu_map.h#L139)

Anyway, it seems odd that the decrease of frequency causes these issues. I mean that laser drivers usually have problems if the frequency is too high, but they never have problems if it is low.

Can you say more about why your laser works worse with 1kHz frequency?

UmkaAP, 2017-07-12:
My laser is TTL. Seen information that at a lower switching frequency each point taken less time, decreases the resolution of the laser in motion.

closed 2017-07-14 by arkypita