arkypita/LaserGRBL #66: Help Wanted

clyde5575, 2017-07-18:
I would like to change the XY Datum from lower left to center of workpiece.  I have tried using the offset but that only moves the lower left position by amount of offset.  Perhaps a custom button?

Thanks

arkypita, 2017-07-19:
If you move the head to the center of workpiece and you send G92X0Y0 command you set a new zero point in grbl in the current position. To be sure that LaserGRBL will report the correct position please read this: #36 

Question: you want to have 0,0 at the center of your table and move to negative space down, left and in positive space right, up. True? Do you want also to use negative space, true?

LaserGRBL only create raster converted pictures with all positive coordinate, but if you open a externally created gcode file with negative numbers LaserGRBL can send it to engraver without problems. The issue is that it crop the graphics, so you can see only the positive part.

This is a circle with 0,0 at the center:
[image]

I think that in the future I could write some code to "autocenter" the preview, to get it working with negative space images.

clyde5575, 2017-07-19:
Thank you for the quick reply, the answer to your question is yes.  I use the laser to augment work I have done on my cnc router. having that feature would enable me to use the same start point to facilitate indexing when moving the workpiece from router to laser.  

Thanks for creating this software, you have done a great job making it the easiest process for converting photos to raster files.

arkypita, 2017-07-20:
I hope you can solve your question with G92 and some math.

I would like to keep LaserGRBL as much simple as possible. Changing coordinate system and reference can confuse a lot of not skilled user, so I think I will never add those features (but never say never).

One things I will do by sure, it is to deal with negative in job preview.

closed 2017-07-20 by arkypita

clyde5575, 2017-07-20:
Now that I understand the way the software is designed it will work just
fine the way it is.  A little math never hurt anyone.

arkypita, 2017-10-08:
last release fix this issue