arkypita/LaserGRBL #67: SD-card version

RunnyBore604, 2017-07-19:
Hi,

sorry for that spam. I hope it will be interested. I created GRBL version that can work without a PC. It requires LCD 1602, SD card module and only one Arduino Nano (not two, like this do others).

https://github.com/RunnyBore604/standaloneGRBL

arkypita, 2017-07-19:
You did a great job. Did you have done some videos of your project at work?

RunnyBore604, 2017-07-19:
One happy user made a few videos about how that controller works: https://cloud.mail.ru/public/3Bjd/jKH9tAcAj

And one video from another person on YouTube: https://www.youtube.com/watch?v=FmmlrxPS5jo

Discussion on Russian forum: http://www.cnc-club.ru/forum/viewtopic.php?f=41&t=16194

arkypita, 2017-07-20:
thanks for sharing

closed 2017-07-20 by arkypita