arkypita/LaserGRBL #68: Add Feature - Edit Imported Raster Image

dcwalmsley, 2017-07-28:
Once image has been added via "Open File" a "Import Raster Image" edit window appears and allows user to create desired output in a GCODE file.  Later the GCODE can be opened in the same manner as opening a raster image.  

I'm requesting the ability to edit the gcode file already saved and re-opened as an image that is seen on the x/y plotter display and allow me to go back into the "Import Raster Image" editor to make the changes and then reapply them as a output GCODE file.

This will allow anyone to run test print and then go back and re-edit the GCODE file using other features and functions as well as resizing the image.  There is no way to resize a GCODE output once created.

hk89, 2017-07-31:
A resize cannot be the only need for this feature. Resizing, rotating, translating, copying into array and spliting the gcode sound like a complete gcode editing method

arkypita, 2017-08-16:
i confirm this is not possible/easy to do

closed 2017-08-16 by arkypita