arkypita/LaserGRBL #69: On-screen Machine and Work position coordinates

hk89, 2017-07-31:
Hi

Sometimes I find it difficult to detect if I am approaching the end of my axes or not. It would be easier to have the machine and work coordinates in hand to avoid tripping a limit switch.

Could it be a feature where these coordinates are being displayed live within the application; maybe at the bottom of the screen next to the overrides. 

Thanks

arkypita, 2017-07-31:
I agree. It will be a new feature on the next release.

Diego

Inviato con AquaMail per Android
http://www.aqua-mail.com

arkypita, 2017-08-19:
added in v2.7.3

closed 2017-08-19 by arkypita

arkypita, 2017-11-07:
[Version v2.8.12](https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.12) can now show working position if different from machine position.