arkypita/LaserGRBL #70: Potential Malware detected. What is this?

jameshudson09, 2017-08-01:
[image]

arkypita, 2017-08-16:
Dear @jameshudson09 

As someone already say my software is sometime reported as suspicius by firewall or antivirus.
This is because of 3 reasons: missing of digital signature, access to internet, needs of admin privilege.

Internet access and admin privilege are needed for "auto update" feature. Digital signature needs a certificate that could be only released by a certification authority under payment.

As LaserGRBL is a free program developed in my spare time and shared with people that can enjoy using it, I will never decide to pay someone that certify my program is not a virus. You trust me? You can whitelist and install LaserGRBL

jameshudson09, 2017-08-16:
arkypita,
  I appreciate the response and all of your work.

thanks for getting back with me.

Best Regards,
James Hudson

closed 2017-08-18 by arkypita

arkypita, 2017-11-21:
@jameshudson09 
Whould you like to test last version? I have changed my installer and now should be more antivirus friendly.
https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.20

jameshudson09, 2017-11-21:
Thanks so much. I will and see if Symantec gives it a false flag.

Thanks,
James