arkypita/LaserGRBL #73: G1 missing

grosseruser, 2017-08-15:
In the attempt to engrave a filled image is missing in most cases the "G1 Fxxx" so that the laser moves with travel speed.
Example:
```
G90
F400
G0 X0 Y0
M5 S255
G1 F100
G91
M5
G0 F400 Y.25
G0 X14 Y11.75 F400 M5
X-.25 Y.25 M3
X-.25 Y.25 M3
G0 X-0.5 Y0.75 F400 M5
G1 F100 X.25 Y-.25 M3
X.25 Y-.25 M3
X.5 Y-.5 M3
G0 X3.75 Y-3.5 F400 M5
X-.25 Y.25 M3
X-.25 Y.25 M3
X-.25 Y.25 M3
G0 X-2.75 Y2.75 F400 M5
X-1 Y1 M3
G0 X-1.75 Y2 F400 M5
X.25 Y-.25 M3
X.25 Y-.25 M3
X.25 Y-.25 M3
G0 X1 Y-1 F400 M5
G1 F100 X.25 Y-.25 M3
X.75 Y-.75 M3
X.25 Y-.25 M3
G0 X2.25 Y-2.25 F400 M5

```

grosseruser, 2017-08-16:
Ok, 
i found it. The bug is in the file "GrblFile.cs" in function "ImageLine2Line".

The following works for me (changes in **):
```
private void ImageLine2Line(Bitmap bmp, L2LConf c){
	bool fast = false;
	List<ColorSegment> segments = GetSegments(bmp, c);
	List<GrblCommand> temp = new List<GrblCommand>();
	foreach (ColorSegment seg in segments) {
		bool changespeed = (fast != seg.Fast); //se veloce != dafareveloce
		if (seg.IsSeparator && !fast) //fast = previous segment contains S0 color {
			if (c.pwm)
				temp.Add(new GrblCommand("S0"));
			else
				**temp**.Add(new GrblCommand(c.lOff)); //laser off
		}
		fast = seg.Fast;
		**//**if (changespeed)
			temp.Add(new GrblCommand(String.Format("{0} F{1} {2}", fast ? "G0" : "G1", fast ? c.travelSpeed : c.markSpeed, seg.ToString())));
		**//**else
			**//**temp.Add(new GrblCommand(seg.ToString()));

		//if (seg.IsSeparator)
		//	list.Add(new GrblCommand(lOn));
	}

	temp = OptimizeLine2Line(temp, c);
	list.AddRange(temp);
}

```
Possibly the "F {1}" part is not necessary.

arkypita, 2017-08-16:
Grbl v1.1 has memories where store the last speed used. when a movement command is sent without specify the speed grbl use the last speed settled, so no Fxxx command needed unless a speed change is required.

So not sending F command is not a bug, is an optimization.
Don't know other g-code controller and older grbl version, but with v1.1 i am pretty sure that LaserGRBL produced g-code work well.

hope @chamnit can confirm

grosseruser, 2017-08-16:
Ok, but with the optimization of the F command, unfortunately, the G1 command was also dropped. My hack helped me, but since I know the code too little, I could not optimally repair.

arkypita, 2017-08-16:
The drop of G1 command is also an optimization. Multiple G command is not needed since gcode is modal and remember the last state.

http://www.cnctrainingcentre.com/cnc-programming/modal-nonmodal-g-codes/

grosseruser, 2017-08-16:
This work fine:
```
G0 X-0.5 Y0.75 F400 M5
G1 F100 X.25 Y-.25 M3
X.25 Y-.25 M3
X.5 Y-.5 M3
```
But in this case simply the G1 is missing and the Laser work wis G0 speed
```
G0 X3.75 Y-3.5 F400 M5
X-.25 Y.25 M3
X-.25 Y.25 M3
X-.25 Y.25 M3

```

arkypita, 2017-08-16:
Ok, I now undertstand what you say!
You are right... I will fix it as soon as possible

chamnit, 2017-08-16:
G1, M3, and F feed rate are all modal. Once set they don't need to be sent again until there is a change in modal state. Also, an F feed rate programmed with a G0 does not alter how a G0 behaves. Some firmwares allow this but it's incorrect. F should only alter feed rate (G1 G2/3 etc). Not rapids rates. But you can override them to run slower via the override commands instead.

arkypita, 2017-08-18:
@grosseruser It seems that the problem is related to the use of PWM-less mode. The reason why I never noticed the problem is that I use the PWM mode, which does not seem to be affected by this bug.

Investigating...

arkypita, 2017-08-18:
@grosseruser  If you want you can check-out and compile my last commit (master) and check if it fix.
Let me know

closed 2017-08-19 by arkypita

grosseruser, 2017-08-19:
I'm sorry but I am on the road with the motorbike in the Pyrenees and enjoy my holidays. I can return at the earliest in 8-10 days.

arkypita, 2017-08-20:
enjoy your trip