arkypita/LaserGRBL #74: Add support for USB camera

alfredanil, 2017-08-17:
Would appreciate if it is possible to connect usb camera with graduated cross wire graphics overlay for work zero measurement and alignment. It can also serve as remote monitor or for time lapse recording. Thank you very much.

arkypita, 2017-08-18:
Hy alfred, it is not a feature that I would like to add in LaserGRBL now or in the feature cause it should be very customized (usb-ethernet, different resolution, drivers...)

Code is opensource, so if someone will add a good camera implementation and submit to me maybe I can merge this feature, but not a priority now for me.

Regards

alfredanil, 2017-08-18:
This feature is already implemented on Grbl Plotter code is available on Github

closed 2017-08-18 by alfredanil