arkypita/LaserGRBL #75: White dotted outlines on image

clyde5575, 2017-08-19:
Down loaded update and now when I edit image for engraving I get white selected areas on photo, must adjust to point of no contrast to eliminate them.  Photo of image and settings are included.  Any suggestions? Thanks

[image]
[image]

clyde5575, 2017-08-19:
additional image
[image]

arkypita, 2017-08-20:
Nothing was added in the very last version that can cause this issue.
What you are showing concern to White Clip, added in June.

The feature of white clip allow you to set a white-point threshold that convert to "pure white" any white value from zero to settled white point.

This could be useful with black and white logos, specially downloaded from web, where sometime whites are not RGB 255, 255, 255 but a very light gray like 254, 253, 254 due to jpg compression or low image quality. In this cases white clip eliminate this noise and prevent laser going burning this areas at very low power wasting time.

With pictures, white clip should be left at zero.

Clipping preview will be shown in pink color when click and drag the "white clip" selector, and disappear when released.

clyde5575, 2017-08-20:
I reinstalled previous version (v2.2.7.2) and recreated exact setup, see
photo pg 3.  As you can see it does not have the white outlines on the
image,so I reloaded v2.2.7.3 and the lines reappeared.  The white clip is
all the way to the left.

arkypita, 2017-08-20:
Maybe I have not understand the issue you are talking about, sorry.
The screenshot you have sent is the final output preview, not the raster import form preview. True?
I am investigating the issue.

arkypita, 2017-08-20:
Thanks @clyde5575 for your fast report of the bug/issue
I have found and fixed it in latest v2.7.4 version

commit 25d34da6 (2017-08-20): fix #75

thanks to user @clyde5575 for fast bug reporting
closed 2017-08-20 by arkypita