arkypita/LaserGRBL #76: Configuration export nit work

gerdvogel, 2017-08-21:
Hi,
I have a small problem.
In order to update the firmware, I would like to save the configuration before and to load it back later.
But when exporting the configuration I get the following error message:

Error exporting config!

it is a bug or a user error?

regards
Gerd

arkypita, 2017-08-21:
Configuration export was one of the first functions I implemented. It's possible that with time it's no longer aligned with the latest code changes. At this time I can only test it with an emulation.

gerdvogel, 2017-08-21:
Hi,

i have testet with grbl 0.9j and 1.1f, with both I gad the sam error.

LaserGRBL 2.7.4

 

regards

Schönen Gruß

Gerd Vogel

closed 2017-08-21 by arkypita

arkypita, 2017-08-21:
Released a new version v2.7.5
Hope it is fixing

gerdvogel, 2017-08-21:
Thanks for the quick help. the bug is fixed. It tested export and import immediately.

 

regards

Schönen Gruß

Gerd Vogel

commit d9f6e940 (2017-08-21): fix for #76

Fix #76 and add emulator config ability