arkypita/LaserGRBL #79: Outlining the job

brakthehun, 2017-08-22:
I would love to be able to see the laser head move to the start of the job,, potentially fire the laser,, maybe at it's dimmest,, and then run a square perimeter around the job so I can see where the job will end up.

It would also be nice to be able to access job variables in the button too. like:

G0X(X_FRAME_MIN-2)Y(Y_FRAME_MIN-2)
M4S255
G1X(X_FRAME_MAX+2)F1500
G1Y(Y_FRAME_MAX+2)
G1X(X_FRAME_MIN-2)
G1Y(Y_FRAME_MIN-2)
M5

So the button would gather the current frame data and compile all things inside parenthesis into to numbers to send the GRBL Controller.

arkypita, 2017-08-22:
I have a blue laser, so I have developed the UI to be readable by me. However I understand your (and other users) problems.

Maybe in a future version I could add some kind of color scheme or color customization.
Thanks for suggestion

arkypita, 2017-08-27:
added in v2.8.0

closed 2017-08-27 by arkypita