arkypita/LaserGRBL #80: Viewing code generated.

StuartB4, 2017-08-22:
Is it just me or is it normal for LaserGRBL to do this.
When I press Start to start the burning process, in the left pane where I see the scrolling GCode list i get a list of 373 lines of M5 (Laser Off) commands before it starts.
Like this:
G90
F4000
G0 X0 Y0
M5 S255
G1 F50
G91
M5
M5
M5
M5
M5
M5
M5
M5....................... 

All the way down to line 343 of the code, and then it starts burning the rest of the code.

M5
G0 X0.05 Y33.85 F4000 M5
G1 F50 Y-.05 M3
G0 X0 Y-0.1 F4000 M5
G1 F50 Y-.05 M3
G0 X0 Y-0.15 F4000 M5
G1 F50 Y-.05 M3
G0 X0 Y-0.1 F4000 M5
G1 F50 Y-.05 M3...............

Would it be possible to add a feature that would store the Speed and Step size slider positions when closing LaserGRBL. So that they are the same settings when restarting it.

arkypita, 2017-08-22:
> Also, Would it be possible to add a feature that would store the Speed and Step size slider positions when closing LaserGRBL. So that they are the same settings when restarting it.

You mean in the positioning/arrows box? Good hint! +1

> When I press Start to start the burning process, in the left pane where I see the scrolling GCode list i get a list of 373 lines of M5 (Laser Off) commands before it starts.

Help me figure out whether it's a generated code issue or just display problem.
If you save the gcode (file-> save) and open it with notepad, do you see the same all the M5 or not?

StuartB4, 2017-08-22:
> You mean in the positioning/arrows box? Good hint! 
Yes, the 2 sliders that go up and down.

>  do you see the same all the M5 or not?
Yes, here's the text file.
[testing.txt](https://github.com/arkypita/LaserGRBL/files/1242643/testing.txt)

arkypita, 2017-08-22:
sound like a real problem but unluckly it does not reproduce in my test :-(
please could you share your testing condition (original image file, raster import setup and image option setup) so i can try to reproduce?

StuartB4, 2017-08-23:
Ok.
1st image is a random image from the internet because it does it with any image I try.
[image]

I used the Crop tool to just get the head.
[image]

This one shows the settings.
[image]

This one shows LaserGRBL running with all the M5 commands n the list.
[image]

arkypita, 2017-08-23:
I am now able to reproduce the issue.
I will investigate as soon as possible

arkypita, 2017-08-23:
No, it does not depend on the grbl version.

StuartB4, 2017-08-23:
Oh, ok. Thanks.

arkypita, 2017-08-28:
Check with last version, it should be fixed now

StuartB4, 2017-08-28:
Ok, will have a look.

StuartB4, 2017-08-28:
Ecellent. Just got the latest version and tried it with the same image and settings as before and it works perfectly.
Thankyou very much.

closed 2017-08-28 by arkypita