arkypita/LaserGRBL #81: diagonal offset during job

alexwht, 2017-08-22:
Hi!
I have a trouble with sporadic diagonal offsets  during job in LaserGRBL 2.7.5. If create .NC file in LaserGRBL and burn this gcode file via UniversalGRBLSender all OK. Please help to resolve trouble.
Thanx.

Im using GRBL 0.9i and ManaSE controller
[image]

arkypita, 2017-08-22:
> I have a trouble with sporadic diagonal offsets during job in LaserGRBL 2.7.5.

Speak explicitly about version 2.7.5 because you did not have these issues in a previous version? I have made a very small change between v2.7.2 and 2.7.5 but a lot of user are reporting problems. Maybe concern this change

> If create .NC file in LaserGRBL and burn this gcode file via UniversalGRBLSender all OK. Please help to resolve trouble.

Does it happen always (and always in the same place) or sporadic?
Scrolling the list of sent commands: are all result OK or some commands has reported ERROR in red?

brakthehun, 2017-08-22:
I have had this happen when my data stream to my Arduino gets interupted.
One thing I've learned with GRBL - use a short USB cable and let the computer be.

alexwht, 2017-08-23:
Thank you for answer. Seems like really problem with USB cable. Error is BAD NUMBER FORMAT, but the GRBL command is absolutly correct. Otherwise, can you add function "repeat command on error", and parameter "count of repeats" in your sender next version? I think that 99.9% same problems with electrical noise and long cables will be resolved. Thank you.

brakthehun, 2017-08-23:
Maybe,, you could add a second Arduino,, bridge the RX and TX and open up the Serial monitor in the IDE. Then you could see what's being sent.

arkypita, 2017-08-24:
@alexwht 
Your suggestion may sound like a good suggestion but it is not praticable. LaserGRBL implements [character-counting streaming protocol](https://github.com/grbl/grbl/wiki/Interfacing-with-Grbl#streaming-protocol-character-counting-recommended-with-reservations) that provides for the subsequent commands to be sent before even having the answer of the previous commands.

So LaserGRBL aggressively sends a series of commands by keeping the grbl queue full, to optimize streming and prevent dead times in communication and execution. Grbl already has the next command as soon as it has finished the previous command, without having to wait for it to be sent.

Repeat on error can only be implemented if sender (LaserGRBL) wait for the complete execution of each single command (whether it is ok or error) before sending next one (or repeating).

So realize what you suggest it need first to change the mechanism by which LaserGRBL stream commands to [simple-send-response](https://github.com/grbl/grbl/wiki/Interfacing-with-Grbl#streaming-protocol-simple-send-response-recommended-for-grbl-v09) than add repeat on error.

Much better that you make some changes to your hardware to make it more solid and robust and error proof. Often and willingly this only means finding the right cable.

Stepper could generate very strong electrical spike on power line that can go back to USB line.
In my experience I have tested 3-4 different cable and only one is 100% reliable. The good one is a samsung phone data cable. Good ones should be short, thick (well shielded) and possibly with ferrite. www.google.it/search?q=usb+with+ferrite

Some other suggestion:

- if you use a laptop plug it to the power line and do not run with battery. doing this you have grounding through main power line.
- place your USB cable far from motors and DC line, to minimize induced noise.
- add a pair of capacitor parallel to DC line, one elettrolitic of 47uF 50v and one ceramic of 0.1uF. this is called "decoupling" and could help in case of electrical noise.

alexwht, 2017-08-25:
Thanks for advices!

closed 2017-08-25 by alexwht

arkypita, 2017-08-25:
wanna try this pre-release version?
https://github.com/arkypita/LaserGRBL/releases/tag/v2.7.6

alexwht, 2017-08-25:
Thanks very much! I'll try next week. Superlight speed development)))

alexwht, 2017-09-08:
A little testing results:
I used bad long (1.5m) USB cable without ferrite and not shielded.
I place 3 smartphones on USB cable and place electrical kettle cable near it for better electrical noise.
There is no problems and no errors with "synchronous" and "repeat on errors" features. 
Perfomance reduced about 5-10% 
Thanx again

arkypita, 2017-09-08:
thanks for feedback